Ultimaker/Cura #7638: [4.6.1 (& before)] Resizing support blockers does so relative to original orientation of the model

KenLucke, 2020-04-30:
**Application version**
4.6.1 (and before)

**Platform**
Macbook Pro OS 10.5.4

**Printer**
Creality CR-10v2 (customized from CR-10s)

**Reproduction steps**
1. open model
2. reorient model
3. apply support blocker
4. attempt to resize support blocker
5. Ugh.

**Screenshot(s)**
githib, for some reason, does not like any attempt by me to upload a file of any type.  Have yet to figure out why.  Amazon is the same way, but every other site accepts my files.

**Actual results**
Axis resizing, either by dragging the bars or typing into the resize fields, is relative to ORIGINAL orientation rather than CURRENT orientation.

 i.e, X-axis (Red) always resizes what was originally the X axis before reorienting.  Y-axis (green) does the same only with the original Y-axis of the model, etc. 

Most confusing, and extremely difficult to overcome if the model has not been simply rotated 90º or 180º in some direction.

**Expected results**
Should resize relative to CURRENT orientation, as does the main model resize.

**Project file**
any file

mahtDFR, 2020-05-01:
It's very difficult to help you with this information. You'll need to provide a screenshot, project file, or log at the very least. I will close this for now, please reopen when you can give us some more information.

closed 2020-05-01 by mahtDFR

Ghostkeeper, 2020-05-06:
See also https://github.com/Ultimaker/Cura/issues/6370. This is a reincarnation of the open issue with rotation and non-uniform scaling: https://github.com/Ultimaker/Cura/issues/4136