Ultimaker/Cura #7639: cura 3.3.1 brim printing inside out when replicating model

jagalexander, 2020-05-01:
**Application version**
(The version of the application this issue occurs with.)
3.3.1

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
win 10

**Printer**
(Which printer was selected in Cura?)
Folgertech FT-5

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)
replicating model 5 time

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)
Brim prints outside in on all but the last model

**Expected results**
(What should happen after the above steps have been followed.)
Brim should print inside out on all models

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

Vandrasc, 2020-05-01:
Hi @jagalexander, we just released 4.6.1, the bug as described by you is fixed in this version (previous versions as well btw), please give it a try and let us know if you still encounter this issue.

Thank you!

jagalexander, 2020-05-01:
I downloaded the latest version last night and it still has the same bug. I can send you the project file if you would like.

Cheers,
Jamie

jagalexander, 2020-05-01:
I downloaded the latest version of Acura last night and it has the same bug. I can send you the project file if you would like.

mahtDFR, 2020-05-01:
Please do. And a log as well.

closed 2020-06-19 by no-response[bot]