Ultimaker/Cura #7641: black project

diego8866, 2020-05-01:
good morning,
when I load the projects under care in the prepare section, the drawings are always seen in black color also change the color of the filament, the problem is solved only using version 3.6, with subsequent versions the STLs are black,
what can I do? thanks

fieldOfView, 2020-05-01:
Can you add a screenshot, and the next time pretty please fill in the issue template?

diego8866, 2020-05-01:
[image]

fieldOfView, 2020-05-01:
Have you pressed that nice blue "Sezionamento" button?

diego8866, 2020-05-01:
in preview the problem does not arise once pressed sectioning I see the slycing correctly, the problem is only in the initial view on the prepare section

fieldOfView, 2020-05-01:
This is how Cura works. You have to slice a model before you can preview how it will print. If you don't want to press the slice button, turn on "Slice automatically" checkbox on the General page of the preferences. You can also install the "Automatic Slicing Toggle" plugin from the Marketplace to quickly toggle that option.

diego8866, 2020-05-01:
ok but in preparation I should see the piece with the color of the filament.
for example in the image the filament is purple, instead the object is black..
I tried to change filament and color but it does not change even after clicking sectioning

mahtDFR, 2020-05-01:
That much is true. Please save a project file (File -> Save...), zip up the .3mf file and attach to this issue so we can try to reproduce it.

If you haven't already, please can you attach a [log file](https://github.com/Ultimaker/Cura#logging-issues).

diego8866, 2020-05-01:
[cura.log](https://github.com/Ultimaker/Cura/files/4563996/cura.log)

diego8866, 2020-05-01:
[1.zip](https://github.com/Ultimaker/Cura/files/4564010/1.zip)

i have only the stl...

mahtDFR, 2020-05-01:
I'm afraid I can't reproduce it from that: 

[image]

If you follow the instructions I posted above to save a project file, I will be able to try with your specific setup. Otherwise we're just guessing.

diego8866, 2020-05-01:
[AA8_1.zip](https://github.com/Ultimaker/Cura/files/4564072/AA8_1.zip)
ok sorry

diego8866, 2020-05-01:
is there a folder where custom filaments are saved?
the origin of the problem is that I had made a black filament but changing the filaments the color remained

it is as if the color setting had been saved and that the program could no longer change it

mahtDFR, 2020-05-01:
Thanks. I have the information to try and reproduce it now. But I can't reproduce it! I think it's a problem with your configuration.

[image]

You could try [resetting your user data](https://github.com/Ultimaker/Cura/wiki/Cura-Preferences-and-Settings-Locations). You can back up your configuration first by zipping the relevant folder if you need to save things.

One side-note: you're using PVA as a build material - is that intended? This is usually a support material. Are you sure you don't want PLA?

diego8866, 2020-05-01:
no pva is selected only to try to change the colour 
i must delete the file of prefferences?

mahtDFR, 2020-05-01:
I can't think why that's happening. As I can't reproduce it here, that would guess there's an error in your configuration somewhere. One way to verify that would be if you zip your configuration file to make a backup. Then delete the original and open cura with out-of-the-box settings to verify if the behaviour persists. If it does, then we know there's an error in your original config file.

If you wish to revert it to how it was, all you need to do is unzip the file again into your settings directory.

diego8866, 2020-05-01:
i removed the files from appdata and localappdata and configured again cura..i had some files of 2 other version of cura, configured again cura but i have no changes..do you know if i can search in other position?

Ghostkeeper, 2020-05-06:
Maybe this is a rendering issue? Like, GPU? Maybe something with the new X-ray in Solid View feature there is something not supported on some graphics cards?

diego8866, 2020-05-07:
THANKS A LOT was only a problem of GPU solved with an update of the driver i have an nvidia geforce GT525M 
nice idea I hadn't really thought about it

closed 2020-05-07 by diego8866