Ultimaker/Cura #7642: Random overextruded layers

hagan66, 2020-05-01:
**Application version**
4.6.1

**Platform**
MacOS

**Printer**
Prusa MK2.5S

**Reproduction steps**
1. Sometimes one or more layers appear overextruded in a print while the rest of the print is perfect. 
2. If I print the same g-code again, the lines are still there. If I change something and reslice they can dissappear or show up att different height.

**Screenshot(s)**

[image]

**Additional information**
I posted about this in the Cura user group on facebook, and other people had observed this aswell.

mahtDFR, 2020-05-01:
That's strange. Maybe you can share a project file with me?

smartavionics, 2020-05-01:
Missing z-steps due to too high jerk/accel or too low stepper current?

hagan66, 2020-05-01:
Hi! 

These two cubes are printed from the same settings, except for infill, which is grid for the bad one, and gyroid for the good one. I can send you the project file shortly aswell.

:-)

/Robert

Skickat från min iPhone

hagan66, 2020-05-01:
Project file and gcode

hagan66, 2020-05-01:
Hi! I tried to repy via email but it did not work with my attachments. I now attach two images of prints printed with exact same settings except for infill, which is grid infill for the bad looking cube and gyroid infill for the good one. Also attached you will find project file and gcode.
[image]
[image]
[Bug Report.zip](https://github.com/Ultimaker/Cura/files/4566029/Bug.Report.zip)

Ghostkeeper, 2020-05-06:
Seems to be a flow rate problem with your nozzle then, to me. I suggest you keep using Gyroid or other infill patterns that don't produce overlap, such as Lines, Zigzag, Concentric or Cross (3D). That's what those infill patterns are there for.

closed 2020-05-06 by Ghostkeeper

hagan66, 2020-05-06:
Hmm, I never had this until the last two versions of Cura.

I will do some checking and get back.

There were actually several people on Cura FB group that also had noticed this recently.

:-)

R

hagan66, 2020-05-07:
> 6 maj 2020 kl. 16:23 skrev Robert Hagström <hagan@me.com>:

I just wanted to say you are probably right, now I got these ugly lines in Prusaslicer as well. So probably not Curas fault :-)

I suspect sticky rods messing up z-movement. I have now oiled them up and printing the same g-code. I will report the result.

R