Ultimaker/Cura #7643: Support structures vary depending on build plate position

evilC, 2020-05-01:
**Application version**
4.5.0

**Platform**
Windows 10 x64  
nVidia Geforce 1080 Ti

**Printer**
Ender 3

**Reproduction steps**
1. Place model on build plate that has overhangs
2. Enable support
3. Slice
4. Translate model

**Screenshot(s)**
Same model, same layer, same settings, same rotation, but different X/Y position on build plate
[image]  

[image]

**Actual results**
Support structure varies depending on build plate X/Y position

**Expected results**
Support structure should be identical regardless of X/Y position

**Project file**
[Brawler Cura bug.zip](https://github.com/Ultimaker/Cura/files/4564889/Brawler.Cura.bug.zip)

Ghostkeeper, 2020-05-06:
Support is not specific to a particular model though. Take for instance this print here:

[image]

If I move the selected object forward, which model would you expect the support to move along with?

evilC, 2020-05-07:
One STL can contain multiple completely separate objects - this is one STL
[image]  
So when you slice, combine everything on the build plate to one logical object, and then calculate the origin point of the supports as the center of this one logical object.
That way, no matter where you place things on the build plate, as long as the relative offsets between them are the same, it will generate identical support

Ghostkeeper, 2020-05-11:
Yeah, that would work.

Maybe it would be better for people that want to move their entire scene to the side of the printer for some reason.

evilC, 2020-05-11:
Well it just seems a little silly as-is
You do a test print, ensure that the supports are sufficient, then add a duplicate of the print to the build plate (Which are nowhere near each other) but you don't get the same supports.
Also, it seems to me that the use-case you posted (two separate models very close together) is the edge-case here (Hell, I don't even know how to get pyramid-like support structures like that - the default options would make those two objects have completely unconnected support structures surely) and the software should not be geared towards that use-case to the detriment of the vastly more common use-case
TLDR multiple identical items on the build plate which are reasonably spaced should have **absolutely identical** support structures by default, anything else seems highly illogical

mahtDFR, 2020-05-12:
We have discussed this and decided to defer it for now due to other priorities.