Ultimaker/Cura #7644: Support structure never connects with model

evilC, 2020-05-01:
I am not wholly convinced this is a bug, but I thought I would report it none the less. It's maybe a case of needing a new setting, or me missing an existing setting

**Application version**
4.5.0

**Platform**
Windows 10 x64  
nVidia Geforce 1080 Ti

**Printer**
Ender 3

**Reproduction steps**
1. Open supplied project
2. Slice

**Screenshot(s)**
[image]

**Actual results**
Support structure never connects with arm

**Expected results**
Support structure should connect with arm (At least periodically)

**Project file**
[Brawler Cura bug.zip](https://github.com/Ultimaker/Cura/files/4564951/Brawler.Cura.bug.zip)

Asterchades, 2020-05-03:
As a broad generality, supports shouldn't come in direct contact with the model. Any time the two come directly in contact they'll fuse together, requiring the supports be cut to be removed - something you can see on occasion if a model has supports on both sides with "Limit Support Retractions" enabled. It does mean that the under surfaces lose some quality but that's the trade-off with having the supports able to be removed cleanly.

You can, of course, configure the supports otherwise - which you'll want to do if you're printing your supports in PVC. Just be aware that if your supports are not soluble than you're going to have to use a blade or clippers to remove them, which is likely to reduce surface quality a lot more than the slight sag that the gap can introduce, as this is going to be like separating the model from itself.

The settings you'll want to adjust (make sure they're visible) are "Support Z Distance" and "Support X/Y Distance". If you have "Support Distance Priority" set to "Z overrides X/Y" than you'll also have a "Minimum Support X/Y Distance" setting. Supports that start on the model also use the "Support Stair Step Height" setting. Note, though, that it's an "all or nothing" situation - you either have all of the support touching or none of it. There's no option to have it periodically touch.

mahtDFR, 2020-05-04:
The above is good advice I think. By changing "Support Distance Priority" to "Z overrides X/Y" I was able to narrow the gap a little. @evilC is this an adequate solution for you?

[image]

evilC, 2020-10-14:
Apologies for letting this languish for so long, but no, this is not a satisfactory solution IMHO
If an overhang is flat, then support interface will be generated and will touch (ish, my understanding is that there is actually a tiny gap, but the floor above will sag to meet the support interface, and so will actually end up touching the interface)
If, however, the overhang is angled, then no support interface will ever be generated.
IMHO what is needed is some setting that ensures that every X layers, support interface will be generated
To clarify - the flat of the top of this part has support interface, whereas the angled part does not, so the material below cannot really be considered "support". Yes, you could alter settings to that the support touches the overhang, but then you would end up with a solid, fused part - the purpose of the support interface is to ensure a partial connection which easily breaks.
What is needed is something to ensure that every now and then, the support under the angled part has support interface which connects with the part, to ensure a proper, removable support
[image]

Asterchades, 2020-10-14:
> the purpose of the support interface is to ensure a partial connection which easily breaks.
I don't necessarily agree with this. My understanding is that the purpose of the interface is to allow you to switch to a different material - such as PVA - for the purpose of printing the very last piece of support, such that the material above either won't stick to it or that the interface can be removed chemically (by water in the case of PVA). That it can also be used with some materials (PETG and PLA in my experience) to form a "solid" top on the supports that still generally breaks away cleanly (akin to a raft) is something of a happy accident.

That being said, I can't reproduce what you're seeing. When I create a similar object (a 10mmx20mm rectangle, tapering up and out 20mm at 45° before a 10mm lip with an additional 2mm on top) I get support interface up the entire sloped side (provided I lower the support overhang angle to under 45°, anyway). Even scaling it to various different sizes doesn't alter that.

eg:
[image]

What I have done, however, is dropped the "Minimum Support Interface Area" to 0. It's likely, and even probable, that those small overhangs are simply not large enough for whatever setting you've got to treat them as needing an interface.

This appears to be made even more pronounced by what appears to be a (relatively) large X/Y offset value which is restricting the supports from getting closer in the first place (I use 0.6mm). in other words, even if you had it producing the interface there it wouldn't be any closer to the object because of the restrictions you've set.

evilC, 2020-10-14:
To be fair, I don't think "I set" those values, IIRC they are what Cura defaults to (Or at least maybe what they used to, and upgrading has kept those values).  
Setting X/Y offset to 0.6 and Minimum support area to 0 does indeed cause it to generate support interface on the sloped area.  
I guess, however, that setting "Minimum support area" to 0 will also cause it to generate support in pretty pointless areas - if this is the case, I still feel that maybe some tweaking to the logic in this area would be worthwhile.  
I guess maybe how it works at the moment is that it generates support for all layers, then goes back and generates the interface - so when it looks at a particular layer, most of the area is support, and only a tiny slice is considered for support interface (ie this area)
[image]
Maybe what instead it needs to do is apply interface on the first pass (ie for the whole layer) and then for the next layer above, change part of that to regular support
ie
[image]
[image]

evilC, 2020-10-14:
Ah, d'oh, I was thinking it was "Minimum support area", not "Minimum support interface area", so yeah, I guess it would have negligible effects.  
Either way, this allows me to work around the issue, so I will close this request. Many thanks for your time

closed 2020-10-14 by evilC

Ghostkeeper, 2020-10-15:
We are not really happy with the state of the "Minimum Support Area" and related settings. The aim (~= reason why we developed it) of the Minimum Support Area is to remove small pieces of support that would get unstable when printing. But in reality it removes too much. The aim of Minimum Support Interface Area is to prevent unnecessary extruder switches for very small parts, but that is of course not an issue for single-extrusion prints.

The default for Minimum Support Area is 0mm² and Minimum Support Interface Area is 1mm². But the Ender 3 profiles that you're using set that to 2mm² and 10mm². The Cura team has little to say about whether that is necessary for that printer or not. The Creality community has already reduced those a few releases ago so I'd assume it's better tuned now.

evilC, 2020-11-05:
@Ghostkeeper apologies for resurrecting this again, but how can I change the minimum support area settings etc that Creality set?
I opened the machine settings dialog, but they are not specified in there
[image]

Ghostkeeper, 2020-11-10:
You'd change it in the normal settings list:

[image]

There is no way in the GUI to change the default (which is indeed the whole point of a default). If you want to change the default for Creality printers, you'd have to edit the [creality_base.def.json](https://github.com/Ultimaker/Cura/blob/master/resources/definitions/creality_base.def.json) file. This is not something we'll support though, so you're kind of on your own there. The setting keys in that file refer to the internal names in Cura. Look up fdmprinter.def.json to find them.

evilC, 2020-11-12:
Excellent, this is exactly what I wanted, I'll set it to 0 in that file and now I don't have to edit it in the settings every time. Thanks!