Ultimaker/Cura #7646: Prime tower processing when "machine_extruders_share_heater" is true

furuiie, 2020-05-01:
I can't speak English but I use machine translation.
I'm worried if it will be transmitted, but I'm looking forward to working with you.

Red: extruder_0   
Blue: extruder_1

In the first layer, when the blue color is not needed, the blue color is printed on the skirt (brim), and unnecessary material is replaced.
[image]
[image]

The first layer should be purging for the blue material, but it is not.
[image]
[image]

The ideal prime tower in the first tier
For red (extruder_0) only [If prime_tower_enable is true]
Do not print around the skirt (brim) with another extruder.
[image]

For two or more colors
Make a prime tower in red and purge the blue inside.
[image]
or
[image]

Added: "value" formula for prime tower position
I think this is better. When "adhesion_type" is "skirt" "skirt_line_count" is ignored. (If the skirt exceeds the width of "brim_width", it will protrude from the plate.)

"prime_tower_position_x": {"value": "machine_width - max(((extruderValue(adhesion_extruder_nr, 'brim_width') * extruderValue(adhesion_extruder_nr, 'initial_layer_line_width_factor') / 100 * (1 if not prime_tower_brim_enable and adhesion_type == 'brim' else 0)) if adhesion_type == 'brim' or (prime_tower_brim_enable and adhesion_type != 'raft') else (extruderValue(adhesion_extruder_nr, 'raft_margin') if adhesion_type == 'raft' else 0)) + (extruderValue(adhesion_extruder_nr, 'skirt_gap') + (skirt_line_count + 1) * extruderValue(adhesion_extruder_nr, 'initial_layer_line_width_factor') / 100 if adhesion_type == 'skirt' else 0), max(extruderValues('travel_avoid_distance'))) - max(extruderValues('support_offset')) - (resolveOrValue('draft_shield_dist') if resolveOrValue('draft_shield_enabled') else 0) - 1" },
"prime_tower_position_y": {"value": "machine_depth - prime_tower_size - max(((extruderValue(adhesion_extruder_nr, 'brim_width') * extruderValue(adhesion_extruder_nr, 'initial_layer_line_width_factor') / 100 * (1 if not prime_tower_brim_enable and adhesion_type == 'brim' else 0)) if adhesion_type == 'brim' or (prime_tower_brim_enable and adhesion_type != 'raft') else (extruderValue(adhesion_extruder_nr, 'raft_margin') if adhesion_type == 'raft' else 0)) + (extruderValue(adhesion_extruder_nr, 'skirt_gap') + (skirt_line_count + 1) * extruderValue(adhesion_extruder_nr, 'initial_layer_line_width_factor') / 100 if adhesion_type == 'skirt' else 0), max(extruderValues('travel_avoid_distance'))) - max(extruderValues('support_offset')) - (resolveOrValue('draft_shield_dist') if resolveOrValue('draft_shield_enabled') else 0) - 1" },

furuiie, 2020-05-01:
I made a mistake in the calculation formula

"prime_tower_position_x": {"value": "machine_width - max(((extruderValue(adhesion_extruder_nr, 'brim_width') * extruderValue(adhesion_extruder_nr, 'initial_layer_line_width_factor') / 100 * (1 if not prime_tower_brim_enable and adhesion_type == 'brim' else 0)) if adhesion_type == 'brim' or (prime_tower_brim_enable and adhesion_type != 'raft') else (extruderValue(adhesion_extruder_nr, 'raft_margin') if adhesion_type == 'raft' else 0)) + (extruderValue(adhesion_extruder_nr, 'skirt_gap') + (skirt_line_count + 1) * extruderValue(adhesion_extruder_nr, 'initial_layer_line_width_factor') / 100 if adhesion_type == 'skirt' else 0), max(extruderValues('travel_avoid_distance'))) - max(extruderValues('support_offset')) - sum(extruderValues('skirt_brim_line_width')) * extruderValue(adhesion_extruder_nr, 'initial_layer_line_width_factor') / 100 - (resolveOrValue('draft_shield_dist') if resolveOrValue('draft_shield_enabled') else 0) - 1" },

"prime_tower_position_y": {"value": "machine_depth - prime_tower_size - max(((extruderValue(adhesion_extruder_nr, 'brim_width') * extruderValue(adhesion_extruder_nr, 'initial_layer_line_width_factor') / 100 * (1 if not prime_tower_brim_enable and adhesion_type == 'brim' else 0)) if adhesion_type == 'brim' or (prime_tower_brim_enable and adhesion_type != 'raft') else (extruderValue(adhesion_extruder_nr, 'raft_margin') if adhesion_type == 'raft' else 0)) + (extruderValue(adhesion_extruder_nr, 'skirt_gap') + (skirt_line_count + 1) * extruderValue(adhesion_extruder_nr, 'initial_layer_line_width_factor') / 100 if adhesion_type == 'skirt' else 0), max(extruderValues('travel_avoid_distance'))) - max(extruderValues('support_offset')) - sum(extruderValues('skirt_brim_line_width')) * extruderValue(adhesion_extruder_nr, 'initial_layer_line_width_factor') / 100 - (resolveOrValue('draft_shield_dist') if resolveOrValue('draft_shield_enabled') else 0) - 1" },

Ghostkeeper, 2020-05-07:
I will use short sentences. This makes translating easier for computers.

> In the first layer, when the blue color is not needed, the blue color is printed on the skirt (brim), and unnecessary material is replaced.

This is not a bug. We intended this. One of the functions of the skirt is to prime the nozzle. This makes the nozzle ready for printing. It ensures that the material is aligned to the tip of the nozzle. This needs to happen for every extruder. Once you're at the blue layers, it will be too late to go back down to the build plate. The print head would collide with the red cube. Therefore, it must prime during the first layer.

The prime tower is intended for dual extrusion. The setting is not visible for printers with 1 extruder.

> The first layer should be purging for the blue material, but it is not.

This is also intended. Cura chooses one material for the prime tower as the most important material. It chooses the material with the best layer adhesion for this. That way the prime tower is strong. This material is used for the outer shell of the prime tower. It is ALSO used for the base layer of the prime tower.

The first layer MUST consist of the same material as the outer shell. Two different materials don't stick well to each other. They shrink at different rates when cooling down. If they consist of different materials, the adhesion won't be as strong. This makes the prime tower topple.

In your case, it chose the red material as the best one. If you use different colours of PLA, the choice is arbitrary. However, if you use PLA and ABS, it will always choose PLA for the outer shell and the base layer.

> When "adhesion_type" is "skirt" "skirt_line_count" is ignored.

Good find! We'll take a look at this. It can indeed go outside of the build volume. This is not only for the prime tower though. This is the case for the entire print! The shadowy border needs to become wider.

rburema, 2020-05-07:
> > When "adhesion_type" is "skirt" "skirt_line_count" is ignored.

> Good find! We'll take a look at this. It can indeed go outside of the build volume. This is not only for the prime tower though. This is the case for the entire print! The shadowy border needs to become wider.

This particular bug is now on our backlog as (internal) ticket CURA-7432 -- Since we're experiencing an increase in number of issues reported, it might be a while before we get to it though.

furuiie, 2020-05-07:
Excuse me. I didn't include the desired picture description.
I think the prime tower brim and outer shell can be left as they are.
It is better to prime inside the outer shell only when using materials other than the outer shell.
[image]

Prime Tower with shared heater (in my case)
If the red block is below, I don't think you need the first layer of blue prime, as it can be processed by the prime tower when using the blue material.
I thought it was a prime tower for that.

If the blue block is below, I print it first with the red material and then the blue around the skirt and brim, but since there is one nozzle, it gradually changes from red to blue.
That distance may not be enough. Therefore, it is necessary to replace it at the Prime Tower.
I set "prime_tower_min_volume" to 40.
"prime_tower_size": {
　　　　"default_value": 30,
　　　　"maximum_value_warning": "35"
},
"prime_tower_min_volume": {"default_value": 40},

[image]
材料の固定位置はノズルの先端から-35 mmです（私の場合）。
なるべく材料の交換は避けたいです。

Incidentally. I think that it is better to print the outer shells of the second and subsequent layers from the inside. This is because the other material gradually changes to the original material outward.

In the case of a shared heater, there are other types, but it is good because you can handle it later with settings etc. There is no way to handle the prime tower.
Please reconsider.
[My environment.zip](https://github.com/Ultimaker/Cura/files/4595604/My.environment.zip)

Ghostkeeper, 2020-05-11:
The prime tower brim is not always enabled though, and also not always sufficient adhesion if it is enabled.

fvrmr, 2022-02-01:
We are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs. 

To be able to focus on the most requested and needed features from our community and from Ultimaker, we have decided to remove this feature request from our backlog. We believe we will not work on this anytime soon and will therefore defer this issue.

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.

closed 2023-07-13 by github-actions[bot]