Ultimaker/Cura #7648: [4.5 and 4.6] orthographic and perspective views are reversed

sk8nfool, 2020-05-01:
**Application version**
Both versions 4.5.0 and 4.6.1

**Platform**
Ubuntu 18.04

**Printer**
N/A

**Reproduction steps**
1. start Cura
2. select either by the **View** menu or by clicking the orthographic/perspective view button in the view box either view

**Screenshot(s)**
the following shows a perspective view but orthographic selected in menu
(Image showing the problem, perhaps before/after images.) 
[image]

the following shows an orthographic view but perspective selected in menu
[image]

**Actual results**
See screenshots above.

**Expected results**
an orthographic view when orthographic view selected and visa versa

**Project file**
N/A

**Log file**
N/A

**Additional information**
By the way - there seems to be a disconnect between the ortho/persp button tooltext and the menu view and the image actually displayed.

Also the perspective view is ... backwards(?). Nearby objects should appear larger than distant opjects (vanishing point construction).

Also zooming with the mouse works differently in each view. In orthographic view the zoom is relative to the mouse position. In perspective view the zoom is relative to the center of the window.

smartavionics, 2020-05-01:
Just measure the lengths (using an old fashioned ruler) of the front and rear edges of your model on the screen images. The top image (orthographic mode) on my screen measures about 82mm across both the front and the back of the model. The bottom image (perspective mode) is approx 135mm across the front and 100 across the back. Where's the problem with that?

smartavionics, 2020-05-01:
Probably one of those "optical delusions" you hear about...

sk8nfool, 2020-05-01:
That'll teach me to eat too many cookies! You are right. They measure the same on the orthographic view and smaller in the back of the perspective view. I'm not sure what to call the view that ISN'T the perspective view but I don't think orthographic is correct. Wikipedia seems to restrict the orthographic view to one of top, bottom, front, back, left or right where all *edges* are the correct length (ignoring scale). Maybe what I was expecting (and what I'm used to looking at) is an *isometric* view. In an Isometric view all horizontal and vertical dimensions are correct but everything else is allowed to be different AND its drawn on 30/60 degree "paper" where 90 degree corners are distorted to 120 degrees or 60 degrees depending of orientation. The main difference *I believe) between isometric and perspective view is that parallel lines merge to a vanishing point in a perspective but remain parallel in an isometric view.

I always thing the *orthographic* view in Cura looks weird. The same is true for openscad.

Feel free to close this non-bug.

fieldOfView, 2020-05-01:
> Wikipedia seems to restrict the orthographic view to one of top, bottom, front, back, left or right

Does it? On wikipedia, what you refer to are call "first or second angle multiview projections". Isometry is another subtype of orthographic projections.

https://en.wikipedia.org/wiki/Orthographic_projection

Ghostkeeper, 2020-05-07:
Rest assured that you're not the only one that made this mistake. At work I even had a hardware engineer telling me that the orthographic projection is backwards and makes far away things larger. That is, until we measured it with a ruler. The human mind is so used to dealing with perspective projections that other projections appear weirdly deformed.

You get used to it though. I use orthographic almost exclusively now.

closed 2020-05-07 by Ghostkeeper