Ultimaker/Cura #7649: Cura 4.6.1 - No Slicing

DevilAle, 2020-05-01:
After updating Cura to the version 4.6.1 I'm not able to slice the models anymore.
Already try to Uninstall/Reinstall.

smartavionics, 2020-05-01:
Unless you fill out the template and provide more info, log files, etc. There's no chance that anybody can help you. Sorry, we're not mind readers.

billmail, 2020-05-01:
I'm having the same problem. 
smartavionic, it's not helpful being sarcastic. You might know this stuff but some of us do not. 
It would be better if you tell us where to find "the template" and where to find the "log files". 
Then we would be happy to provide the needed information.

DevilAle, 2020-05-01:
**Application version**
4.6.1

**Platform**
Windows 10 (Home) 

**Printer**
ANYCUBIC MEGA ZERO

**Screenshot(s)**
[image]

**Actual results**
Nothing happens..

**Expected results**
Slicing the model.

**Project file**
[image]

billmail, 2020-05-01:
Following DevilAle's lead...
My setup is 4.6.1 version
Creality Ender3
Windows Pro 64bit.
All previous versions of Cura worked just fine. Updated to 4.6.1 and now it no longer slices.
Completely uninstalled including all profiles (I have backups) and reinstalled but 4.6.1 still will not slice even the simplest files that I have sliced in the past.

smartavionics, 2020-05-01:
Log files ?

billmail, 2020-05-01:
Since you didn't say where to find the log file, I took a guess and found the attached.
Is this what you wanted?
[cura.log](https://github.com/Ultimaker/Cura/files/4566131/cura.log)

smartavionics, 2020-05-01:
Yes, that's a good file. Hopefully the Cura devs will see something in there that gives an indication of where the problem lies. I don't use Windows so know nothing about it but do you use a firewall? I think they have caused problems similar to this before now. Might be worth turning the firewall off and seeing if that makes a difference.

billmail, 2020-05-01:
That was the problem. 
I've tweaked the firewall and now it is slicing as it should. Don't know what part of the update was tripping up the firewall, but at least it is working again.
Thanks

DevilAle, 2020-05-02:
After adding Cura Unimaker to the ''Exceptions'' list of my firewall the Slicing tool works as supposed!
Thank you for your help!

mahtDFR, 2020-05-04:
Good to know you have a solution for this. Closing.

closed 2020-05-04 by mahtDFR