Ultimaker/Cura #7651: [Feature Request - Extensions/gcode modifiers/retraction settings]

KenLucke, 2020-05-01:
I'd like to see a script for the gcode modifier post-processing extension that would allow changing retraction settings (at least on/off, but preferably some of the other major ones) at specified layers or heights.  This could be added into the "Change At" extension or a standalone.  

There are times when I need different retraction settings for different parts of my model. For instance, I have one model that has a bunch of tiny, separate details on the first few layers with nothing to hold them in place, and retraction tends screw these up, and I end up canceling the print like 4 out of 5 times.  But the rest of the model is normal, and the retraction is necessary to eliminate other issues.  My solution now is to save the gcode one way, modify the settings, save it again, then splice/merge) the two files together at the necessary layer.  Royal PITA.

mahtDFR, 2020-05-07:
Thanks for raising this. We discussed the issue this morning. Although we see some use for your proposal, we are not able to prioritise it right now, so it has been deferred.
If someone in the community would like to work on it, please make a pull request.

Ghostkeeper, 2020-05-07:
Actually, I'm seeing now that it has already been implemented by a pull request we merged two weeks ago: https://github.com/Ultimaker/Cura/pull/6778

So this will be in Cura 4.7. Lucky!

Ghostkeeper, 2020-05-07:
[image]

closed 2020-05-07 by Ghostkeeper

KenLucke, 2020-05-07:
> On May 7, 2020, at 04:33, Ghostkeeper <notifications@github.com> wrote:
> 
> 
> Actually, I'm seeing now that it has already been implemented by a pull request we merged two weeks ago: #6778 <https://github.com/Ultimaker/Cura/pull/6778>
> So this will be in Cura 4.7. Lucky!
> 
> 
> 

LOL.  :)