Ultimaker/Cura #7653: Wrong Velocity, Acceleration (and Jerk ?) control

walterbux, 2020-05-02:
**Application version**
4.6.1

**Platform**
Windows 10 Intel HD Graphics

**Printer**
Custom printer
I have 2 different custom built cartesian printers similar to Prusa I3: one derived from Anet A8 light and quick and another one with leadscrew axes driven by NEMA 23 motors bigger and slower.

**Reproduction steps**
1. I slice an object of about 50 x 50 x 15 mm with Cura and it is perfecly printed by the Anet A8 derived printer whilst in the other one the X and Y motors lose steps.   
2. I have configured lower speeds and lower accelerations to solve the problem on the second biggest printer but it seems that the changes in speed and accelerations have no effect on the slicing. I say more: if I leave default speeds and accelerations the motors lose less steps (it is better).
The biggest printed has done a successful speed test with 120 mm/s print and 600 mm/s^2 acceleration without losing steps.
The same printer had successfully printed many objects with cura versions < 4.0 !

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
A setup solving the problem, a bug correction, an explanation...

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

[Tappo azzurro-cura461.zip](https://github.com/Ultimaker/Cura/files/4567885/Tappo.azzurro-cura461.zip)

Ghostkeeper, 2020-05-07:
You have disabled acceleration control, and thus Cura will not control acceleration in this print. I can see this in the settings that you've overridden which are listed at the bottom of the g-code:

```
;SETTING_3 e = draft\\n\\n[values]\\nacceleration_enabled = False\\nadhesion_ext
```

Please enable the Acceleration Control setting. It should put M204 commands in the g-code then that control the acceleration rates.

If I'm wrong about this, please share an actual project file rather than your g-code. With a project file I can reproduce a slice and see what Cura is doing to produce it.

closed 2020-05-07 by Ghostkeeper

walterbux, 2020-05-08:
You are right, I have left accelerations disabled because the printer was losing less steps in this configuration.
Now I send you both project and compiled files in a simplified environment: I have created a custom printer with the same setups but without complex start and end gcodes as well as 1 extruder instead of 6 because there was no color change and only the default extruder was used.

By the way inside the Gcode no M204 appear.

Best regards.

Walter
[tappo.zip](https://github.com/Ultimaker/Cura/files/4599067/tappo.zip)

walterbux, 2020-05-09:
After a check I see that in Repetier Gcode M204 has the following meaning:

M204 - Set PID parameter X => Kp Y => Ki Z => Kd S<extruder> Default is current extruder. NUM_EXTRUDER=Heated bed

so nothing to do with acceleration.
In Repetier flavor the expected commands should be:

M201 - Set max acceleration in units/s^2 for print moves (M201 X1000 Y1000)
- M202 - Set max acceleration in units/s^2 for travel moves (M202 X1000 Y1000)

and 

M220 S<Feedrate multiplier in percent> - Increase/decrease given feedrate for speed...

Best regards.

Walter

Ghostkeeper, 2020-05-12:
Ah, it's Repetier. So that seems to be working properly then. This is at the start of your g-code:
```
;LAYER_COUNT:62
;LAYER:0
M107
M202 X600 Y600
;MESH:covid_mask_cap.stl
G0 F2400 X158.953 Y115.03 Z0.3
M201 X300 Y300
;TYPE:WALL-INNER
```

Its sets the print acceleration to 300,300 and the travel acceleration to 600,600, like your settings indicate.