Ultimaker/Cura #7654: Support z distance, não funciona

pumberto, 2020-05-02:
**Application version**
4.61

**Platform**
Windos 10
**Printer**
Anycubic chiron
Tevo tornado

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

pumberto, 2020-05-02:
Distance from support is not working

closed 2020-05-02 by pumberto

Ghostkeeper, 2020-05-07:
Seems to be working fine on my end.