Ultimaker/Cura #7656: Duplicate inside wall line when printing a wall of thickness 3x line width

tturpin, 2020-05-02:
**Application version**
4.6.1

**Platform**
Ubuntu 18.04

**Printer**
Creality Ender 3

**Reproduction steps**
Slice the attached model (the top of a box) with 0.4mm line width
[top_fn60.zip](https://github.com/Ultimaker/Cura/files/4568570/top_fn60.zip)

**Screenshot(s)**
This is how the top layer is sliced (other layers are similar) :
[image]

**Actual results**
Two inside wall lines are generated. Only one is visible in the screenshot, but the animation shows that the middle wall line is printed twice, one time before each outside wall. The second time it is printed with almost no extrusion (0.16mm for the full perimeter, looking at the Gcode). This is not only a waste of time, but it also reduces the pressure in the nozzle, which cause under-extrusion at the start of the next outside wall.

**Expected results**
Only one middle wall should be printed (possibly with a larger line width if the thickness is only slightly greater than 3x line width).

Another issue is that the behavior is not very predictable : it depends on the resolution of rounded corners for the box. For example setting "$fn=36" instead of 60 in the OpenSCAD model causes the center to be printed as infill instead, which is not necessarily a big problem, except that in two of the corners, the line surprisingly becomes a very narrow skin pattern with a ridiculously small line width:
[top_fn36.zip](https://github.com/Ultimaker/Cura/files/4568592/top_fn36.zip)

[image]

While experimenting, I have seen other strange behavior : sometimes identical layers are sliced differently (either with the middle as infill or as two inside wall lines), and I have even seen one side of the box treated differently from the rest of the perimeter in the same layer (I didn't actually checked in the STL that the walls always have the same thickness, but from the OpenSCAD code, they should).

jarekdudzinski, 2020-05-03:
I have the same problem - Cura 4.6.x on MacOS with Ender 2.

mahtDFR, 2020-05-04:
Please save a project file (File -> Save...), zip up the .3mf file and attach to this issue so I can try to reproduce it.

tturpin, 2020-05-04:
This is the project file for the "duplicate inside wall case"
[CE3_top_fn60.3mf.zip](https://github.com/Ultimaker/Cura/files/4574571/CE3_top_fn60.3mf.zip)

And this is the one with lower resolution of the rounded corner, sliced with only outside walls and infill, with some very thin parallel lines in two corners.
[CE3_top_fn36.3mf.zip](https://github.com/Ultimaker/Cura/files/4574573/CE3_top_fn36.3mf.zip)

mahtDFR, 2020-05-04:
Thanks for the project files. I can reproduce the weird corner behaviour and can say this is a known issue with gap filling. There's a fix for this in progress.

[image]

Not sure I understand the extra wall behaviour, but this could also be related.

tturpin, 2020-05-04:
> I can reproduce the weird corner behaviour and can say this is a known issue with gap filling. There's a fix for this in progress.

Good to hear ! Is there already a git issue number for that ? Any idea of when the fix might be available (possibly as a development branch snapshot) ?

jarekdudzinski, 2020-05-04:
> Not sure I understand the extra wall behaviour, but this could also be related.

Check how the printhead moves when printing 3xline walls:
1. first it prints middle wall (yellow)
2. prints inner wall
3. NOW is this problematic extra-pass of printhead over middle wall
4. prints exterior wall

Ghostkeeper, 2020-05-07:
Cura will always generate only contours around the shape of that layer. So if your shape has a big hole in the middle, it'll also generate a wall around that hole. So you will always have an even number of walls.

There are currently two exceptions to this:
* Wall Overlap Compensation can reduce the line width of one of the walls if they overlap (3 < x < 4 line widths). There is also a minimum flow rate for that, so if the line width gets reduced to less than e.g. 50% it'll use a travel move instead.
* Gap filling if there is an empty space (2 < x < 3 line widths). This produces actually just a single line. Except that this also has problems.

We have a research topic going on right now that would solve this: https://github.com/Ultimaker/CuraEngine/pull/1210

tturpin, 2020-05-07:
Thanks for the explanation. I don't know how what the branch you mention does but I think it would useful to be able to remove this "even number of walls" rule.

I have tried the minimum flow rate setting, using 5% turns the duplicate wall into a travel move, maybe a default value other than 0% would be sensible here ? Of course, it would be even better to detect that the whole line has been replaced by a travel move, and remove it completely.

Also, I understand that flow equalization could be an alternative, but it doesn't seem to have any effect here. Is it supposed to work on inner wall lines ?

Ghostkeeper, 2020-05-07:
> maybe a default value other than 0% would be sensible here ?

Yeah, myself I'm using like 50% if I see this problem in one of my prints. Below 50% of the nozzle size it really doesn't extrude consistently any more. You get little balls of plastic instead of a solid line.

> Also, I understand that flow equalization could be an alternative, but it doesn't seem to have any effect here. Is it supposed to work on inner wall lines ?

No, strangely that is only applied to the gap filling technique, not to compensated walls. See my (very brief) explanation of this setting on the [Settings Guide](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/speed/speed_equalize_flow_enabled.md). I think they found that for walls it produces too many vibrations which make the visual quality worse, but this was a long time ago.

tturpin, 2020-05-07:
> Yeah, myself I'm using like 50% if I see this problem in one of my prints. Below 50% of the nozzle size it really doesn't extrude consistently any more. You get little balls of plastic instead of a solid line. In the Ultimaker profiles they didn't change this yet because the material and process engineers haven't yet tested the results thoroughly.

OK, I will use that, and maybe give the development branch a try...

photographer-mememe, 2020-06-03:
When I have ring (cylinder with a hole in the middle) with thickness of 1.5mm, with 0.5mm line width, I expect 3 lines, but I get a similar 4th line.  I assume because of the 5% wall overlap.  With 3 walls, it means there is a 0.05mm gap in the inner wall.

I followed the recommendation on this forum post:

> "Turn on the wall overlap compensation and set the min wall flow to, say, 40 (not really very critical)."

https://community.ultimaker.com/topic/28634-cura-insisting-on-two-inner-walls/
This actually does prevent the 4th line for me.  But the 0.05mm gap will still be there.  It would be nice if there was a "wall gap compensation and max wall flow" setting that does the opposite of the overlap feature, by increasing the flow and shifting the line a little bit over so that a wider line will fill in the gap up to the max flow.

Hurly26, 2020-06-28:
Just found this thread. I'm experiencing the same problem and wanted to share a [Reddit thread](https://www.reddit.com/r/3Dprinting/comments/hh5a1p/cura_wall_bug_cura_can_only_handle_wall/) that I started on the issue as well in case it helps provide additional context. The main video used for the post may not be super helpful but I posted a second video as well in the comments which (I believe) fully explains the problem.

skumrask, 2021-05-09:
Was just about to report same bug, but found this here. I have the same issue under 4.9.0. I attach here an even simpler project which uses standard profile. And a screenshot of a preview where one can see the printhead going over the middle wall line a second time. This is under macOS 10.15.7.

[BB1_Glass.3mf.zip](https://github.com/Ultimaker/Cura/files/6447484/BB1_Glass.3mf.zip)

[image]

Ghostkeeper, 2021-12-14:
Fixed in https://github.com/Ultimaker/CuraEngine/pull/1210.

closed 2021-12-14 by Ghostkeeper