Ultimaker/Cura #7657: Create  a setting to specify which video adapter to utilize to avoid opengl issues with docks

wrpetersen, 2020-05-02:
**Is your feature request related to a problem? Please describe.**
When my laptop is docked to my displayport dock, I have to disconnect it before opening Cura to allow it to use my embedded graphics. I can reconnect the dock as soon as the application is open and use it without issues. Otherwise it automatically tries to use the dock for processing which throws the "opengl" driver error.

**Describe the solution you'd like**
I would like there to be a setting to permanently specify which graphics device to utilize in the Cura settings menu. That way when you open the application it will automatically point to my embedded graphics rather than the dock adapter.

**Describe alternatives you've considered**
At startup, the application can check if there are multiple graphics devices when it can't find the opengl driver on the device it is defaulting to.

**Affected users and/or printers**
(Who do you think will benefit from this? Is everyone going to benefit from these changes? Or specific kinds of users?) I had a hard time figuring out what the issue was. I assume other users check the forums. Most of the forums are describing general OpenGL driver issues and direct you to reinstall. I had researched a long time because i7 processors do have OpenGL and finally it clicked that it was because I had multiple video devices. I would guess that less savvy people would give up and not use the application. It is quite a hassle to undock your laptop just to open the application especially when windows decides to go through an update overnight and close out all of your applications.

**Additional context**
(Add any other context or screenshots about the feature request here.)

Ghostkeeper, 2020-05-07:
Honestly this is more in the domain of your operating system rather than something in Cura. I've never seen any preference in any other 3D application either, like games or CAD or something. For instance Blender, OpenSCAD, CloudCompare, etc.

The operating system has methods for this though, e.g. Bumblebee on Linux. Windows has it built in I think.

fieldOfView, 2020-05-07:
@wrpetersen, could you try unchecking the option "Restore window position on start" in the General preferences? This way Cura should always open on your primary screen, which I would think is using the integrated gpu.

wrpetersen, 2020-05-07:
When you have a closed laptop on a dock, the primary screen is one of your monitors on the dock. It doesn't help. The issue is that as a software product, the average person will not understand what is going on and assume their machine doesn't have the correct hardware for the application. I personally know the issue and have a workaround but I have seen multiple forums where people are just stumped. FreeCad does the same thing on my machine so I agree that other 3D applications encounter this issue. OS qwerk or not, having an application check for multiple GPUs to avoid errors preventing its use would be an improvement in the overall robustness of that application. I would think it would be a beneficial enhancement overall.