Ultimaker/Cura #7658: [4.6.1] Bridge Skin Direction changes if you move the model on the plate

thelastWallE, 2020-05-02:
**Application version**
4.6.1

**Platform**
Windows 10

**Printer**
Ender-3

**Reproduction steps**
1. Enable "Bridge Settings/Bridge has multiple layers" under Experimental
2. Goto preview and on the layer with the first bridge skin layer
3. Move model/models on the plate
4. Slice
5. Repeat step 3 and so on

**Screenshot(s)**
[image]
[image]
[image]
[image]

**Actual results**
When i just move the model on the print bed, the direction of the first bridge skin is changing direction with random behavior.
One time it's in direction x-axis, another time it's in direction y-axis and also sometimes with an angle(maybe 45° or 135°, if it takes the normal skin direction setting for this).

**Expected results**
The first bridge skin layer should have the shortest path from the wall to the opposite wall.

**Project file**
[2x1x1 brick.zip](https://github.com/Ultimaker/Cura/files/4568673/2x1x1.brick.zip)

**Log file**
Should not be relevant

**Additional information**
Found this with multiple models and tried it after that with only one model on the bed.
Another thing: Would be also nice if the first skin line wouldn't start so close to the existing wall line. You can recreate that if you play with the bridge skin density. Maybe i will start another issue for just that?

smartavionics, 2020-05-02:
That's a really interesting bug you have found there (I think). You can verify if I am right or not as to the cause of the lines changing direction if you can make a very small change to your brick model. If possible could you please shift that central pillar a tiny amount or change its shape so that the centre of its bounding box no longer coincides with the centre of the bounding box of the outer walls. You probably need to offset the bounding box centre by approx 0.1mm to be sure. If you can do that and check if the skin directions no longer alter when you move the brick about it will verify where the problem lies. Thanks.

smartavionics, 2020-05-03:
Don't worry about doing that test as I have verified the bug's existence here. It's a very ancient bug that long predates the arrival of the experimental bridge settings. What's happening is that the ancient code tries to determine the best direction of the skin by finding the regions on which the skin is supported (the 'islands') and then calculating the angle of the line that joins the centre of the two largest islands that support the skin. For your object, the two islands are the outer walls of the brick and the central column. The problem arises because the centres of those two islands are almost exactly in the same location (within +/- 1 micron). When you move the object on the buildplate, the centres of the islands can shift very slightly and so the calculated angle changes. I think it would be better in this situation for the code to not calculate an angle and let the normal skin angle for that layer be used (i.e. +/- 45 degrees, or whatever). I shall submit a PR that stops this behaviour.