Ultimaker/Cura #7659: Bridge detection does not turn fan back off

rc-fire, 2020-05-02:
**Application version**
4.6.1

**Platform**
MacOS

**Printer**
Creality CR-20 Pro

**Reproduction steps**
1. Set Fan Speed to 0%
2. Activate "Enable Bridge Settings"
3. Set Bridge Fan Speed to 100%
4. Leave "Bridge Has Multiple Layers" disabled

**Actual results**
The printer prints without fan and sets fan to 100% for first bridge. Then the fan stays on 100% for the rest of the print.

**Expected results**
The printer prints without fan and sets fan to 100% only for bridges. After the bridge is completed fan is turned back off.

smartavionics, 2020-05-02:
Hi, could you please provide a project file that shows this bug. Do File -> Save and then zip or rename the .3mf and attach to this issue. Thanks.

rc-fire, 2020-05-02:
Here is a file. I had to recreate and cannot test it right now since my printer is busy. Acutally I noticed my Description was a little incorrect. The fan was turned on for all the infill, it still turned off for the shell. 

[ProjectFile.zip](https://github.com/Ultimaker/Cura/files/4569347/ProjectFile.zip)

rc-fire, 2020-05-03:
Ok obviously it was my vault as the Bridge Skin Support Threshold was set to 100% which means that all infill is considered bridge. I still would call it a minor bug since the description clearly says " less then this percentage" and in my world 100% is not less then 100%.

I let the admins decide if this shall remain open.

smartavionics, 2020-05-03:
The blurb for that setting says "If a skin region is supported for less than this percentage of its area, print it using the bridge settings." Well, in your model, the skin is supported for less than 100% of its area because it's only supported around the outer edges and by the 3 circular pilars so I would expect it to be treated like a bridge skin.

Ghostkeeper, 2020-05-07:
It is treated like a bridge skin. You can't see the difference in that project file normally but if you change the Bridge Skin Flow to 10% you'll see that it classifies that entire layer as bridge skin. That's correct because roughly 14% of it gets supported by the three pillars, which is less than 100%.

The edge case of 100% being supported doesn't exist because if 100% of a bridge would be supported, there would not be a bridge. It'd just be somewhere in the middle of your model. The edge case of 0% does exist though, if you have a piece floating in mid-air. However bridging won't save you then. Other percentages are not really a concern since it's comparing areas down to square micrometres, so if you have a percentage of like 60% it would have to be exactly 60% down to the micrometre, for which I don't think anyone really cares what happens in that case. It's not what you are seeing.

The bug still stands though. However I can't reproduce it even with your project file. I see that it activates the fan when it starts bridging using `M106 S255`, but then it deactivates the fan when it's done bridging using `M107`.

GregValiant, 2024-12-15:
The post-processor "Advanced Cooling Fan Control" (included since 5.7) when set to "By Feature" will turn the fan on for all sections of gcode that are labeled with ";Bridge".  When the next feature is encountered (EX: 'TYPE_FILL') then the fan will go to the set speed for the feature.
I'll go ahead and close this.  Let me know if that is premature.

closed 2024-12-15 by GregValiant