Ultimaker/Cura #7664: Allow to enforce --single-instance via config file

reibuehl, 2020-05-03:
**Is your feature request related to a problem? Please describe.**
Opening new instances of Cura 4.x takes longer and longer with each version. Fusion 360 seems to not use the --single-instance command line parameter when starting Cura and the default file association that is created by the Cura installer also seems to not use it.

**Describe the solution you'd like**
It should be possible to enforce the use of --single-instance even if not specified on the command line by setting a config file setting and/or preference that ensures that --single-instance is always used when starting Cura. This would reverse the default behavior of creating a new instance each time. 

**Describe alternatives you've considered**
Wrapping the Cura.exe in a batch file would be a workaround but would require additional changes to the default file associations that are not easy to do for end users.

**Affected users and/or printers**
Most users of Cura would benefit from this behavior that was the default in earlier versions of Cura like 15.04.

**Additional context**
<none>

fieldOfView, 2020-05-03:
I have looked into this before. The reason why this is not an easy fix is that in the current architecture it takes a little while before the application preferences are read. By the time the preferences are fully loaded Cura is already well underway starting up a new instance. Perhaps an environment variable could help, but that is not easily user-settable in preferences.

reibuehl, 2020-05-03:
I understand... is there a technical reason why multiple instances became the default in the first place? I think it used to be the opposite in old versions of Cura. 
An environment variable would at least be easier to set for user - compared to modifying the registry for the file association and would also help with Fusion 360 AFAIK.

Ghostkeeper, 2020-05-07:
No, it's always been like that. Here's a screenshot of 3 instances of Legacy Cura running at the same time:

[image]

reibuehl, 2020-05-07:
Maybe on Linux...? On Windows, when I opened a STL in legacy Cura and then another one, it got added to the build platform of the existing instance.

Ghostkeeper, 2020-05-07:
Maybe the operating system implemented that then, because Cura didn't, until the `--single-instance` parameter was [added](https://github.com/Ultimaker/Cura/commit/0e306df1bcc0c0e6126530a28a498073be85898c) in Cura 2.5. Neither Legacy nor modern Cura.

fieldOfView, 2020-05-07:
We can talk long or short about whether the feature existed or not, but here's a PR that works around the problem of the preferences system not yet existing early in the boot process: https://github.com/Ultimaker/Cura/pull/7708

PR #7708 (2020-06-02): Add a preference to use a single instance for loading files
This PR adds a preference to use a single instance for loading files which has the same functionality as the --single_instance command line argument. ~It adds a very minimal preference parser that specifically looks for this particular preference instead of spinning up the full preference singleton, so the early boot is not delayed by much for this feature.~ Edit: the preferences are already available at that point, so I can use them as is.

The "single instance" feature is commonly used by users of CAD applications that launch Cura f [... cut]
closed 2020-06-02 by nallath