Ultimaker/Cura #7665: [4.6.1]

sk8nfool, 2020-05-03:
**Application version**
4.6.1 and possibly other earlier versions

**Platform**
Ubuntu 18.04

**Printer**
N/A

**Reproduction steps**
1. Start with '.../Ultimaker_Cura-4.6.1.AppImage Stand_Link.stl" where ... is path to Ultimaker file

**Screenshot(s)**
Note the "Invalid File" error at bottom of screen
[image]

However just reopening the file works. Note that Cura seems to understand the current directory correctly but not on startup.
[image]

**Actual results**
See above

**Expected results**
file to open

**Project file**
N/A

**Log file**
Logs (cura.log, stdout.log and stderr.log) don't mention open failure or file name

**Additional information**
It appears that the appimage is messing with Cura's idea of the current directory. This has been a chronic problem but not annoying enough to really be called a "bug" .... but it is a bug.

smartavionics, 2020-05-03:
Then why not do .../Ultimaker_Cura-4.6.1.AppImage ~/Stand_Link.stl to make the filename path absolute?

smartavionics, 2020-05-03:
Sorry, that assumes the stl is in the home directory, but the point is, use an absolute path name rather than a relative one for the stl file.

smartavionics, 2020-05-03:
Personally, I wouldn't call it a bug unless the Cura docs actually state that it can open files specified on the command line relative to the launching directory.

sk8nfool, 2020-05-04:
I understand that documentation sets the ground rules BUT an unqualified
pathname lives in a context and that context needs to be considered unless
overridden. I believe that Linux documentation states that un-rooted
pathnames are relative to the current working directory. I also believe
that Windows documentation says the same thing. So I believe that this
situation is a bug. This is expecially true because it appears that Cura
assumes all un-rooted paths start at the home folder. This simply isn't
true.

Now, all that being said, I don't know what the API for AppImages is nor do
I know what the Windows API looks like.

smartavionics, 2020-05-04:
> I believe that Linux documentation states that un-rooted
pathnames are relative to the current working directory. I also believe
that Windows documentation says the same thing. So I believe that this
situation is a bug.

Yes, but any program is free to change its current directory. There's no rule that says that they can't.

sk8nfool, 2020-05-04:
Absolutely, but if it chooses to change it's working directory it needs to
remember the context from which it was invoked. This is why I said I don't
know what the AppImage API looks like. In particular how that context is
handled. I do know that the appimage is mounted on a dynamically created
mount point and the application is run from there but I don't know if the
working directory at the time of invocation is maintained in any way. I
will know this the next time I respond.

smartavionics, 2020-05-04:
> Absolutely, but if it chooses to change it's working directory it needs to
remember the context from which it was invoked.

Sure, it could do that but, AFAIK, it's not obliged to.

smartavionics, 2020-05-05:
Interestingly, the Cura startup code already contains the following...
```
# GITHUB issue #6194: https://github.com/Ultimaker/Cura/issues/6194
# With AppImage 2 on Linux, the current working directory will be somewhere in /tmp/<rand>/usr, which is owned
# by root. For some reason, QDesktopServices.openUrl() requires to have a usable current working directory,
# otherwise it doesn't work. This is a workaround on Linux that before we call QDesktopServices.openUrl(), we
# switch to a directory where the user has the ownership.
if Platform.isLinux() and hasattr(sys, "frozen"):
    os.chdir(os.path.expanduser("~"))
```

So Cura's current directory should be the user's home directory.

Also, AppImage sets the OWD environment variable to the current directory so it would be possible to chdir to that although I would have thought that the current directory would maybe need to be owned or writeable by the user running Cura?

Ghostkeeper, 2020-05-07:
The current directory doesn't necessarily need to be owned by the current user. It's perhaps good practice to put executables in a read-only directory. Windows does this with its Program Files folder for instance.

See also conversations in https://github.com/Ultimaker/Cura/issues/6194 and https://github.com/Ultimaker/Cura/commit/f4dc9cc6a2fa290461409cafc84a35a1e8bc6c65 that led to that change. Apparently it's something in Chrome that it writes a temporary file to the CWD when the OS is told to open a URL with Chromium as the default application for https. If that file can't be written, it fails to open the page.

mahtDFR, 2020-05-08:
After some discussion we have decided to defer this. We think it's a valid feature but we're not able to prioritize it right now. If anyone external would like to try, please make a pull request.

sk8nfool, 2020-05-08:
Sorry for not responding. I went off to learn a little about AppImages - not an easy task. As it turns out the appimage mechanism tried to help applications deal with the environment changes. The page [here](https://docs.appimage.org/packaging-guide/environment-variables.html) describes how to work around these changes. In particular the appimage mechanism (I don't remember if it is the runtime or the AppRun that is the culprit) creates an environment variable, *OWD*, that is set to the Original Working Directory in effect at the time the appimage was invoked. So the code snippet from Cura's startup code should try changing to whatever value is in environment *OWD* first. If that doesn't work change to the value found in the variable *HOME*. I'll try to do this as @mahtDFR suggests.

smartavionics, 2020-05-08:
Yeah, I mentioned OWD, 3 days ago....

sk8nfool, 2020-05-09:
So you did. Missed that. I don't understand the problem with QT so I can't speak to that but in general, with both Linux and Windows, you only need write permission on a folder if you want to create files/folders in that folder. Ownership may be involved in determining write permission but isn't required per se. The same is true for the file itself. Write permission is only required if the user wants to save any changes. Of course search permission is required to find the file and read permission is required on the file itself in order to read the file. But you already know this.
By the way, at least with Linux, read permission on a folder/directory is only required if one wants to list the contents of the folder/directory. It is not required to **use** the folder/directory in a path. For that one needs **search** permission.

sk8nfool, 2020-05-09:
ok - I can't seem to build Cura so I have to withdraw my offer to try creating a patch for this. It's probably moot in any case since I can only test on Ubuntu 18.04.

smartavionics, 2020-05-13:
The PR to provide a fix for you has been rejected and will not be in future UM Cura releases. My own releases still feature this change so if it's important to you to have this, you could install my Cura which you can find at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0. Please read the README.md file there before using.

sk8nfool, 2020-05-13:
Mark,
Thanks much. An initial test seems to work just fine.
Dave (the skating fool)

sk8nfool, 2020-05-14:
It's not so much that its important to me its that the current behaviour is
inconsistent with the rest of the applications I use on Linux.

Ghostkeeper, 2020-05-18:
I do think that behaviour is incorrect. We can't safely change it though, without reintroducing other bugs. A workaround can probably be found, but the part of our userbase that runs Linux is small to begin with.

sk8nfool, 2020-05-18:
At least you agree that it's bad behaviour. I guess the diagnostic
("Invalid File") should be changed to something like "Can't Open File" or
something. Cura can't say the file is invalid when it can't even open it.
Other tools find the file quite valid and so does Cura once its lead by the
hand to find the file.

Ghostkeeper, 2020-05-20:
Yeah that's a good suggestion. I've applied a change here for the next version: https://github.com/Ultimaker/Uranium/commit/94de63848d469f94667d49ba2888b00a13d2490e

closed 2023-06-24 by github-actions[bot]