Ultimaker/Cura #7666: [4.6.x] printing models where thickess is composed of odd number of passes

jarekdudzinski, 2020-05-03:
When slicing model with wall thickness of odd number of passes (1,5 mm wall with 0,5 nozzle - it gives 3 lines in the wall) - Cura adds one more move, as it would print a wall, but without extrusion. I need to set "Wall line count" to 3 - as other parts of the model require this.

This generates problems, because after this move, less material is extruded on the beginning and it's visible as underextrusion on the exterior wall.

It looks exactly like this, when printing some cilindrical model:
1. middle wall is printed
2. inside wall is printed
3. this movement without printing occurs
4. exterior wall, which begining is underextruded

jarekdudzinski, 2020-05-03:
Please remove - just noticed issue beeing already reported

fieldOfView, 2020-05-03:
You can close the issue yourself. I can't by the way, or I would do it for you.

jarekdudzinski, 2020-05-03:
Oh, ok - never used github before :)

closed 2020-05-03 by jarekdudzinski