Ultimaker/Cura #7667: [4.6.1] Can't change Nozzle size in Machine Settings (i want 0,8 but it's always 0,4 - not saved and resetting to 0,4)

psyche386, 2020-05-03:
**Application version**
4.6.1

**Platform**
Windows 7 pro x64
Nvidia GeForce GTX 780

**Printer**
TronXY x5sa-400

**Reproduction steps**
Goto "settings" - "printer" - "Manage Printers" - "Machine Settings" - "Extruder1" - and try to change Nozzle Size (to 0,8 for example) - then press "Enter" or "Close".
Then repeat and see that Nozzle Size has not changed.

**Screenshot(s)**
https://sun9-22.userapi.com/c858436/v858436817/1e213f/dDWzI3EQBjQ.jpg
https://sun9-14.userapi.com/c858436/v858436817/1e2149/qeGTAp6AaaM.jpg
https://sun9-66.userapi.com/c858436/v858436817/1e2153/5SGE_oCq8eI.jpg
https://sun9-20.userapi.com/c858436/v858436817/1e215d/i2HuWHwqJVY.jpg

**Actual results**
Nothing changed

**Expected results**
Wanna change nozzle size

**Project file**
No project needed, bug is in the other place

**Log file**

**Additional information**
This bug a was seeing in the version before 4.6.1 but it was floaty, but in version 4.6.1 it grown up to stable

fieldOfView, 2020-05-03:
Try using a comma instead of a dot.

psyche386, 2020-05-03:
Thank you, fieldOfView! it solved the problem.
Must i close the issue? Sorry, i'm nube here.

psyche386, 2020-05-03:
> Попробуйте использовать запятую вместо точки.

> Try using a comma instead of a dot.

Thank you, fieldOfView! it solved the problem.
Must i close the issue? Sorry, i'm nube here.

psyche386, 2020-05-03:
> Try using a comma instead of a dot.

Thank you, fieldOfView! it solved the problem.
Must i close the issue? Sorry, i'm nube here.

fieldOfView, 2020-05-04:
Yes, please close the issue; it has been reported before

closed 2020-05-04 by mahtDFR

Ghostkeeper, 2020-06-11:
(Since we neglected to say which issue this is a duplicate of, please refer here: https://github.com/Ultimaker/Cura/issues/6546 )