Ultimaker/Cura #7668: [4.6.1] Inconsistent support generation on features like counterbores; Even vs Odd millimeter depths

jcwebber93, 2020-05-04:
**Application version**
4.6.1

**Platform**
Windows 10, Windows 8.1

**Printer**
Custom FFF, Ultimaker S5 (fresh install)

**Reproduction steps**
1. Enable support on second (or extruder other than main)

**Context and Test Setup**
Sorry, this one is a doozy!

Most of the details below apply to "CFFFP_5Y-Axis.Bearing.Housing.420mm.x.465mm.Enclosed.IDEX.3mf" **only**. The second project file exhibits many of the same properties, but the pattern varies somewhat. Nonetheless, the behavior is not consistent. 

I loaded in a bearing holder of sorts with the bearing hole center line perpendicular to the build plate. After slicing, I saw that there was insufficient support and interface material inside the "counterbore" for the bearing. At this time, the counterbore depth was 5mm. I did some fiddling with it in Cura to see what was going on (some scaling tricks, etc), but set up a test layout where I varied the depth from 6 to 15mm, and then a variation of that where the through hole was removed (leaving only the recess.

The issue can be seen, in varying degrees of severity, with a large combination of settings. 

**Screenshot(s)**
Indicated feature of concern
[image]

Test setup - models grouped by columns have identical counterbore depths, the top row is without the through hole, bottom row is. The bottom right are some miscellaneous variations. 
[image]

First layer. Already the issue can be seen. A, C, E, G, and I (even values for depth (in MM)) are starting correct, with B, D, F, and H (odd values for depth (in MM)) stating with decreased support material. Regarding "J-MOD," this is a 15mm depth item, which would normally show the decreased support. However, as a test I raised it .2mm off the build plate. In this case, the support generates normally, *but this is not consistent* across item depths (some instances need to be raised .6mm).
[image]

Many more layers into the build. As expected, the top row is generating support normally (in the entire recessed region), with the corresponding even models generating support over the overhung areas that need it. 
[image]

Detail view of set A. Support and interface generating normally
[image]

Detail view of set B. Support in the full counterbore piece is NOT generating correctly. 
![6](https://imgur.com/XD90SgK.png

Elevating the model slightly as seen in J-MOD can work, but the value is not consistent. For 15mm depth, .2mm was sufficient, however for other variants .6mm was needed.
[image]

**Actual results**
Support and interface is not generated consistently based on these countebore depths. "UMS5_1eY-Axis.Bearing.Housing.420mm.x.465mm.Enclosed.IDEX.3mf" exhibits many of the same issues as "CFFFP_5Y-Axis Bearing Housing^420mm x 465mm Enclosed IDEX.3mf", however most of the testing was done with the latter project.

**Expected results**
Support and interface should generate such to support the future build material that will run parallel to the build plate.

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)
[UMS5_1eY-Axis Bearing Housing^420mm x 465mm Enclosed IDEX.3mf.zip](https://github.com/Ultimaker/Cura/files/4572375/UMS5_1eY-Axis.Bearing.Housing.420mm.x.465mm.Enclosed.IDEX.3mf.zip)
[CFFFP_5Y-Axis Bearing Housing^420mm x 465mm Enclosed IDEX.3mf.zip](https://github.com/Ultimaker/Cura/files/4572376/CFFFP_5Y-Axis.Bearing.Housing.420mm.x.465mm.Enclosed.IDEX.3mf.zip)

**Additional information**
Files have ZIP append to the end, remove the .ZIP to load the files into Cura.

Ghostkeeper, 2020-05-07:
This is a very well-written and well-researched bug report. Kudos!

But I'm afraid it's also a duplicate of a bug that we've seen before, and consider one of the more serious bugs of CuraEngine (probably the most serious bug in support generation).

To workaround the bug, set the Support Stair Step Height to 0. For this model there are no negative consequences to that, since the support doesn't rest on any sloped surface here.

I'll close this as a duplicate of https://github.com/Ultimaker/Cura/issues/6379 .

closed 2020-05-07 by Ghostkeeper

jcwebber93, 2020-05-08:
> This is a very well-written and well-researched bug report. Kudos!
>
> But I'm afraid it's also a duplicate of a bug that we've seen before, and consider one of the more serious bugs of CuraEngine [...]

Thank you for the update! Will use that workaround in the future. Thanks!