Ultimaker/Cura #7673: "Select a Printer" menu failing

gwgorman, 2020-05-04:
**Application version**
(The version of the application this issue occurs with.)
4.6.1

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
MacOS 10.15.4

**Printer**
(Which printer was selected in Cura?)
Unable to select a printer

**Reproduction steps**
1. Settings > Printer > Add Printer
2. Select "Add non-networked printer"
BUG>> "Ultimaker S5" is selected and you cannot scroll the window

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
Screen does not scroll

**Expected results**
Able to scroll to select

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

evtrados, 2020-05-06:
Hello @gwgorman, 
Thanks for reporting the issue you are facing. I tried your reproduction steps using Cura v4.6.1 on MacOS Catalina (10.15.4), but could not reproduce the issue. Printers are always being added with no issue for me and scrolling is possible too.
Would it be possible to zip your configuration folder (you can access it by going to Help --> Show Configuration Folder) and send it to us? Maybe something else is causing it. 
Thank you!

gwgorman, 2020-05-06:
here you go. I am using a laptop with a docking station and 2 external monitors in a [monitor][laptop][monitor] configuration. I have Cura on the left monitor. Wondering now if maybe it's that....will undock sometime today & see if that changes anything. 

[Archive.zip](https://github.com/Ultimaker/Cura/files/4586984/Archive.zip)

Ghostkeeper, 2020-05-07:
Using that configuration folder doesn't reproduce it for me, but I'm using Linux anyway so the interface widget bugs are bound to behave a bit differently. It adds an Ender 3 printer already, and going to the menu to add a new printer just allows me to scroll fine.

evtrados, 2020-05-15:
I also cannot reproduce it neither on Windows 10 nor on MacOS Catalina using your configuration folder. Did you try un-docking the laptop to see if it still occurs @gwgorman ?

gwgorman, 2020-05-15:
@evtrados just did the undock-test, menu works perfectly when un-docked. Re-docked, put Cura on the left monitor where it failed before, now it works perfectly. No idea why....closing for now. Thanks everyone for also checking. This "no travel" lock-in keeps my Mac docked a lot more than it used to!

closed 2020-05-15 by gwgorman

evtrados, 2020-05-15:
Indeed, I can imagine! :)
Nice to hear that it works for you now!