Ultimaker/Cura #7674: Coasting/ recoasting

sanczess, 2020-05-04:


mahtDFR, 2020-05-04:
> Processing an issue will go much faster when this is filled out, and issues which do not use this template WILL BE REMOVED and no fix will be considered!

Before filing, PLEASE check if the issue already exists (either open or closed) by using the search bar on the issues page. If it does, comment there. Even if it's closed, we can reopen it based on your comment.

Please fill out the issue template.

Ghostkeeper, 2020-05-07:
That is normal if you use too much coasting. Reduce your coasting amount. Coasting depletes the nozzle by oozing instead of printing a line. As always with oozing, the oozed material is lost and manifests itself as underextrusion just afterwards.

Try Retraction Extra Prime Amount to compensate.

closed 2020-05-07 by Ghostkeeper