Ultimaker/Cura #7676: Supports not changing

kyman97, 2020-05-04:
Support pattern cannot be changed. When I change the support pattern to Lines from Zigzag, it does not actually change in the preview mode or on the actual file itself. This goes for all of the support patterns. Cura 4.6.1 seems to be stuck on Zigzag mode even though other support patterns are selected, the print even prints with zigzag supports even though that is not what is selected. Please urgently fix this for the tabletop 3D printing community will definitely catch this and will flood more with issues to be fixed about it. Thank you.

rburema, 2020-05-06:
I can't reproduce this.

Can you _please_ fill out the template? Maybe there's something in the logs.

kyman97, 2020-05-06:
I don't understand what you mean

Ghostkeeper, 2020-05-07:
We mean that you should fill out the template that appears when you create a new bug report. It asks for information like which printer you are using, and asks for a project file, among other things. [This guy](https://github.com/Ultimaker/Cura/issues/7696) did that properly. Thank you Sinedied!

closed 2020-05-07 by Ghostkeeper

kyman97, 2020-05-07:
I filled out exactly what popped up for a bug report, alright? Don't give
me that crap.

Ghostkeeper, 2020-05-11:
No you didn't. This it the template that pops up when you click on "Bug report": https://github.com/Ultimaker/Cura/blob/master/.github/ISSUE_TEMPLATE/bug-report.md

In your case, we are missing:
* Platform
* Printer
* Screenshot(s) (although for this issue that is really not necessary)
* Expected results
* Project file
* Log file

Since the reproduction steps don't work to reproduce your result, we're asking for more information. For instance, it could depend on the platform or the printer you're using. It could depend on the individual settings you've changed (hence a project file) or if we can't reproduce it at all maybe there is a hint in the log file that would allow us to track it down anyway.