Ultimaker/Cura #7677: On Ultimaker 2+ bed lifts up into nozzle before moving to the starting point of a new print causing nozzle to be immersed in extruded material. This happened originally with Cura 4.5 but was resolved with 4.6. It has now returned with 4.61.

achillesmjb, 2020-05-04:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

fieldOfView, 2020-05-04:
Please fill in the issue template instead of putting your life's story in the subject line.

achillesmjb, 2020-05-04:
Wow that’s rude. F.U.

nallath, 2020-05-06:
It might be, but despite the bug template being very, very clear in what we want / need to solve issues, people still fail to read it (and subsequently make us do a lot more work to figure out what needs to be done). Don't you think that's a bit rude too? 

So, unless the template is filled in, we simply won't handle a ticket. We have no other choice.

achillesmjb, 2020-05-06:
A mistake is just that. A response to a two sentence explanation being “my life story” is rude. Seems to me the response just needed to state that the template must be filled out in order for a ticket to be handled. Nothing rude there and to the point.

fieldOfView, 2020-05-06:
> A response to a two sentence explanation being “my life story” is rude.

For the record, I did not mean to offend you.

achillesmjb, 2020-05-06:
Thank you. I will certainly be sure to fill out the form correctly in the future. 

Mike

Ghostkeeper, 2020-05-07:
Looking forward to seeing that! Be sure to mention what material it's dipping into, because it seems to me that there should be no material before it moved to the starting point of a new print? This printer does no prime blob.

I'll close this issue then as promised in the issue template.

closed 2020-05-07 by Ghostkeeper