Ultimaker/Cura #7679: [3.6.1] PETG profile defaults wrong

LordMelkor, 2020-05-04:
**Application version**
3.6.1

**Platform**
Win 10 x64

**Printer**
Ender 3 Pro

**Reproduction steps**
Looked at Generic PETG material settings.

**Actual results**
Temperatures look like a copy/paste of the PLA temps which are probably quite a bit too low for PETG.

**Expected results**
Different settings than Generic PLA when selecting the Generic PETG material.

rburema, 2020-05-06:
There as no such thing as a 3.6.1, as we never needed a patch for that version. Can I assume this is a typo and you're referring to the most recent version, 4.6.1?

In any case, generic PLA 1.75mm and generic PETG 1.75mm are _different_ at least (200 vs 215 degrees Celsius), but 215 definitely seems too low.

However, if I go back to stock 4.5, 4.0, or even 3.2 (not Ender 3 on that last mentioned version), it's _all_ 215 for generic PETG 1.75mm, so while it seems _wrong_ it doesn't seem _different_ to me.

I'll bring it up with my colleagues regardless.

mahtDFR, 2020-05-07:
@LordMelkor Thanks for raising this. We discussed this morning and decided that since there's no way of testing this for us (we rarely use PETG and only use 2.85mm filaments), we won't be making any changes to this profile.

If you would like to modify the profile and have good results, and have tested it thoroughly, then please make a pull request to us and we will consider merging it.