Ultimaker/Cura #7680: Save as 3MF project file doesn't save project with links to underlying STLs

sidhant, 2020-05-05:
**Application version**
4.6.1

**Platform**
Windows 10

**Printer**
Ultimaker 2+

**Reproduction steps**
1. Load models
2.  Add support blockers and a bunch of custom tedious settings
3. If the underlying model changes, these changes can be brought into Cura by Right click> Reload All models. Very convenient.
4. Save the whole setup as a project file .3mf
5. Close Cura
6. Load .3mf file. Everything loads as when saved. Good.
7. Make changes to underlying STL (like step 3), pressing Reload all models does nothing. 
8. Redo all the work in 1 and 2 unfortunately.

**Actual results**
On loading the 3MF, the link to original STL files are broken and lost.

**Expected results**
A "project file" should save the state of the project and the links to underlying assets as they were at the time of saving. It should not save a static and absolute version.

fieldOfView, 2020-05-05:
> A "project file" should save the state of the project 

I fully agree. Having said that, this issue has been reported before and dismissed.

To relieve your pain a little bit, the MeshTools plugin from the Marketplace adds the functionality to the right click menu to replace individual files with a file on disk. Not quite as convenient as having a working "reload all meshes", but it will let you change your project

sidhant, 2020-05-05:
@fieldOfView Thank you for the prompt response. I have the MeshTools plugin and on selecting right click>replace, it wrecks havoc. It imports TWO copies of the model, one which is for some reason scaled so large it does not fit my screen, and the other as if the STL has been freshly dropped. Deleting the weird scaled copy leaves behind the correct model, **however** the original model is still there. "Replace" means, to remove and put another -- not add another. So that is a bug I need to file with MeshTools.

Finally, with the above MeshTools behavior, when I delete the original model, Cura deletes all Support Blockers and the tedious custom settings that had been put in. Back to square one of the original issue that the work needs to be redone.

My hack has been to leave Cura open with my work until I think I am done making changes.

fieldOfView, 2020-05-05:
> it wrecks havoc

Can you please open an issue on the MeshTools issue tracker? It would be nice if you could add your project and the model file you use to replace, so I stand a chance to reproduce and subsequently fix the havoc.

> when I delete the original model, Cura deletes all Support Blockers 

That is how support blockers work; they are "parented" to the original model, so if you remove the original model the support blockers have no parent and go away too.

fieldOfView, 2020-05-05:
See also https://github.com/Ultimaker/Cura/issues/3232 and https://github.com/Ultimaker/Cura/issues/3201

sidhant, 2020-05-05:
Looking through the 3MF schema, it appears that the format is designed to capture meshes and their metadata such as materials, etc. It is supposed to capture meshes on its own and not as a link ref to an external mesh such as a STL file.

I can see why this was dismissed as there is no real way to include this in 3MF (I think).

What is **highly misleading from a UX perspective is that Cura calls exporting the current mesh setup to a 3MF format, _as saving a project_.** IMHO, those are not the same things! It should simply be called "Export to 3MF" and should not be called Save Project. Well OK, I hear ya.. it saves slicer settings as well. So maybe, we should reconsider and have the underlying STL link be saved too?

fieldOfView, 2020-05-05:
> there is no real way to include this in 3MF (I think).

Sure there is. You can add all the metadata you want to 3MF files.

sidhant, 2020-05-05:
> > there is no real way to include this in 3MF (I think).
> 
> Sure there is. You can add all the metadata you want to 3MF files.

I stand corrected. I just saw that its basically a zip and the Cura/ folder has whatever settings you want, so possibly the filename could be saved and loaded back. When the underlying file does not exist, the "reloadAll()" action defaults to re-reading the 3MF mesh.

> Can you please open an issue on the MeshTools issue tracker? 

Yes. Will do.

sidhant, 2020-05-05:
> That is how support blockers work; they are "parented" to the original model, so if you remove the original model the support blockers have no parent and go away too.

Yes and I was hoping the MeshTools would be smart enough to replace the parent and not create a new parent/new node.

fieldOfView, 2020-05-05:
Like I said, please open an issue in the MeshTools issue queue and I'll see what can be fixed. Replacing the "parent" mesh should leave the support blockers in place. Removing the parent mesh will not.

sidhant, 2020-05-05:
> Replacing the "parent" mesh should leave the support blockers in place. Removing the parent mesh will not.

Correct! And my expectation was such from MeshTools (and from Cura that it will maintain underlying links to STL).

fieldOfView, 2020-05-05:
A development snapshot of MeshTools is now available that fixes the issue in that plugin:
https://github.com/fieldOfView/Cura-MeshTools/issues/11

Ghostkeeper, 2020-05-07:
> Sure there is. You can add all the metadata you want to 3MF files.

True. However you are not allowed to omit the 3D data. If you link to a different file in the 3MF file, you'll have two files which could both be modified and out of sync, or only the 3MF data is modified, or only the STL data is modified. Reloading is inherently ambiguous then.

Reloading a 3MF file (without fieldOfView's MeshTools plug-in) is currently intended to behave the same as reloading an STL file: It reloads the 3D data in that file. If you load a few STL files in Cura and save the build plate as a new STL file, you would also not expect the new STL file to change if you then change one of the original files it was created from.

sidhant, 2020-05-07:
> Reloading is inherently ambiguous then.

That's an excellent point. I agree that in light of your argument, it is NOT a good idea to link to a STL via 3MF.

I think this is coming down to the "Save Project" language being a UX failure. It should not be called a project. That parlance in software packages has a long running meaning of saving the exact state of a project including preservation of all functionality. Cura should call it "Export as 3MF package". Just like saving a STL is *never* called saving a project in any sane software package.

Ghostkeeper, 2020-05-11:
"Export as 3MF package" would also be problematic though, because there is also an "Export" function which saves as 3MF by default. The 3MF file there is an actual 3MF file. Cura's project files put extra data in there that isn't supposed to be in a 3MF file.

MIOsystems, 2022-01-04:
Please reconsider this issue because i also think this would be a great benefit for the use of project files. In my case i use many (different) parts in one build that are precisely placed on the build plate. When making updates on parts i need to reposition all those parts again.

I would suggest to add a source link to the meta data and use that when reloading. When opening the project, the project mesh can be used.  Surely its possible that the source is not there anymore but its easy to popup an user request to look for the files or just ignore them.

dracode, 2022-01-14:
> Please reconsider this issue because i also think this would be a great benefit for the use of project files.

I agree.  This could be done with zero changes to the file format, even, by making it a semi-manual process.  the .3mf (zip) file already contains the "3dmodel.model" file inside of it, which in turn keeps a reference to the original model's file name: 
`./3D/3dmodel.model:		<object id="2" name="Frame Curve Fit Test.stl" type="model">`

It doesn't save a complete file path, but that's okay -- we can prompt the user for that information.

I propose adding a new menu option, "Reload Model From Specified File" or something, which will produce a pop-up window, identical to the current "File -> Open File(s)..." dialog box, that simply loads the selected model in place of the existing one, preserving scale/rotation/position/etc transformation information, just as the current "Reload All Models" feature does.

fieldOfView, 2022-01-14:
> I propose adding a new menu option, "Reload Model From Specified File" 

The MeshTools plugin, available from the MarketPlace, adds that functionality (when right-clicking in the viewport).