Ultimaker/Cura #7682: Freezes when slicing

Klestch653, 2020-05-05:


Klestch653, 2020-05-05:
Cura hangs when cutting layers. The indicator can stop in the middle, at the beginning, or at the end - no difference. The program was demolished, cleaned the registry and all files, too. Also disabled the Firewall - nothing. Please do not answer with complex sentences, as I am average in English. Thank you in advance.

**Application version**
4.6.1

**Platform**
Windows 7 i5 2300

**Printer**
Custom FFF printer (My Sidewinder X1 is not available)

**Screenshot(s)**
[image]

**Project file**
All objects

**Log file**
[DxDiag.txt](https://github.com/Ultimaker/Cura/files/4580390/DxDiag.txt)
[cura.log](https://github.com/Ultimaker/Cura/files/4580394/cura.log)

rburema, 2020-05-06:
From the logs I can see the firewall is _not_ the problem, as the _connection_ to the engine is juts made fine. It seems it's killed (for example about a second after it started) after it started from Cura (frontend).

Do you do anything _while_ it's slicing?
 - Changing the scene/model
 - Ctrl-P

Otherwise it might be that your mouse registers multiple clicks when clicked once, then, the engine is scheduled for stop almost as soon as it starts (which it might not immediately).

Klestch653, 2020-05-06:
Hi. Yes, I did it, but it doesn't help

Ghostkeeper, 2020-05-07:
> *Project file*
> All objects

Could you send a project file anyway, with like a simple cube or whatever? Perhaps CuraEngine is getting into an infinite loop because of your settings.

Schuif67, 2020-05-16:
I keep trying Cura after  the 3.6 version , but every "update"makes the program unfortunately worse, now it wont even start

Bbop-3D, 2020-05-18:
Cura does not really freezing at slicing but it looks like. In some cases it takes about 5 Minutes for the indicator to move on. Lower Versions were 10 times faster at slicing.

Ghostkeeper, 2020-05-19:
Please send a project file. It allows us to reproduce the slice so that we can see what's taking so long to slice. For me it slices most of my prints in a few seconds.

Bbop-3D, 2020-05-19:
This one takes 6:41 at my i7-860@3,5GHz to slice...
[bowl_1_x3_011512_456.zip](https://github.com/Ultimaker/Cura/files/4651895/bowl_1_x3_011512_456.zip)

Ghostkeeper, 2020-05-22:
@Bbop-3D I think your issue is different than Kletch653's. His log file indicates that the slicing data didn't get received by CuraEngine, and so the slice doesn't properly start, but the progress bar does get put on a random progression (still don't know why that is random).

Your complaint seems to be just that it takes a long time to slice.

My best bet is that some firewall system is blocking Cura from making connections through a local socket. We didn't change anything about that local socket connection in the last 4 years, but some virus protection systems are a bit overzealous about one program talking to another on your computer.

Ghostkeeper, 2020-05-22:
@Bbop-3D Here are my measurements of the slicing times with two setting alterations you could do, which in your case probably won't impact the quality or printing time much:
* Your project file: 153.4s
* Disabled Compensate Wall Overlaps: 98.4s
* Infill Pattern "Lines": 143.5s

closed 2020-06-19 by no-response[bot]