Ultimaker/Cura #7683: Estimated Time remaining

RowdyRocket, 2020-05-05:
I usually load up Cura to see about where I am on my print to get an idea of how much time is remaining. 

It would be wonderful if the slider in preview mode would give you an estimated time remaining from the point selected in the preview slider bar.

[image]

fieldOfView, 2020-05-05:
Next time please fill in the issue template. I'm guessing you use the OctoPrint plugin?

The biggest problem here is that for performance reasons, Cura normally does not display gcode files, but specially created layer data. That layer data is no longer available when a new slice is made, let alone when Cura is restarted. There is also no direct correlation between gcode line numbers and layer data, which makes it hard to sync up the information about print progress with the layer data.

RowdyRocket, 2020-05-05:
Sorry I didn't use the template. And No, no OcyoPrint plugin, that was just something I threw together in a paint program to show what I was looking for. Does Octoprint offer this? I'm not asking that it track in real-time just an estimated time as I scroll through the preview. Simple math time divided by slices times slice I'm currently viewing should tell elapsed time then subtract that from overall time to get remaining.

Thanks for the quick response. I'll use the template next time.

Ghostkeeper, 2020-05-07:
CuraEngine could send over the print time estimates along with the layer data. It already calculates this (and puts it in the g-code). I don't really see much of a problem with that. This is different from the g-code-line-number information in that other feature request, because there is just a single point where the layer gets sent as a whole to the front-end, so it's not sensitive to programmer mistakes or hard on maintenance.

The only problem that I see is where to fit it in the interface. For most people there is a big panel of settings there, which is why we moved the layer counter to be displayed above the handle. It would also be a bit ugly/overwhelming to have panels coming out from multiple directions, in my opinion. I'm no expert in GUI design though, so maybe my colleagues have a better idea.

mahtDFR, 2020-05-08:
We think this is a valid feature but we have decided to defer it as we can't give it priority right now. We would be very open to someone contributing a solution as a pull request.

louiswouters, 2020-06-08:
@RowdyRocket You may use [this](https://github.com/louiswouters/3D-Printing/blob/master/scripts/ShowProgress.py) PostProcessing script to write the current layer and estimated time to the display of your printer.

5axes, 2021-01-22:
> The only problem that I see is where to fit it in the interface. For most people there is a big panel of settings there, which is why we moved the layer counter to be displayed above the handle. It [...]

In some other Slicer displaying the Slicing time is not a big problem and doesn't necessary an "Ugly" panel
[image]

Nico04, 2022-08-18:
Any update on the this feature?

Vandresc, 2022-08-18:
As far as I see in the history of this issue, this ticket was deferred, meaning it will stay open but we do not have time to address it any time soon. As always, we are open to a Pull Request.

Flying-Roger, 2023-01-23:
I am also looking for such feature.
As a workaround, I open the gcode file in an editor and search for "TIME_ELAPSED:" to browse through the layers, but it would be nice to have this time displayed in the Cura preview, together with the layer index.
To prevent complexification of the interface, why not "simply" add a second line (with the elapsed time) under the layer # in the cursor as shown below?
As the total printing time is already visible in the information window, showing also the remaining time is not necessary.
[image]

GregValiant, 2025-12-09:
The post processor "DIsplay Info on LCD" includes this capability using M117 (to the LCD) and M118 (to a print server like Octoprint).  In addition it includes a "fudge factor" to tweak Cura's estimate closer to reality.
Is that sufficient and can this be closed?

Flying-Roger, 2025-12-11:
If I understand correctly, Cura cannot display the time information because this information is only available in the back end.
However, the front end can display the total time, meaning that a time information is passed back to the front end.
Make it an array (indexed with the layer number) and the problem is solved.
OK, maybe easier to say than to do ;-)