Ultimaker/Cura #7685: there is something in CREALITY CR-10S Pro g-code the Problem is: when it start to print something it draw a triangle then it print the thing

kak121, 2020-05-05:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

fieldOfView, 2020-05-06:
Please fill in the issue template. The subject line is not the place to put your whole issue description.

You may be interested in this forum thread: https://community.ultimaker.com/topic/31676-how-do-i-turn-off-this-initial-triangle/?tab=comments#comment-259620

rburema, 2020-05-06:
I agree with @fieldOfView , it's nearly impossible to diagnose problems if we don't have these reports.

However, this does sound like something that could've been added later to the start gcode or as a post-processing script. What is your start gcode and is ther an icon with tools next to the slice-button area?

fieldOfView, 2020-05-06:
@Rburema, the linked forum post contains a link to a pr that fixes this.

For now, disabling active bed leveling on the printer is rumored to be a quick fix

rburema, 2020-05-06:
It seems that fix was merged into master 2 weeks ago. This means it'll _probably_ be fixed in 4.7, if enough people try the beta of that version in about a month or so and/or the fix just works.

rburema, 2020-05-06:
Let's actually refer to the PR: https://github.com/Ultimaker/Cura/pull/7526

Ghostkeeper, 2020-05-07:
Fixed in the next version then.

closed 2020-05-07 by Ghostkeeper