Ultimaker/Cura #7686: Trouble printing g code!! Please help!

aralas77, 2020-05-06:
**Application version**
CURA 4.6.1

**Platform**
WINDOWS 10 64 BIT)

**Printer**
ENDER 3 PRO EXTENDED (ADJUSTED SETTINGS IN CURA 400*400*500)

**Reproduction steps**
Hello Everyone!, this is my first time posting on Github so I apologize in advance if my questions seem vague so please bear with me. Im having a hard time Printing g code files correctly so any help will be appreciated. I recently designed a small clip on fusion 360 , saved as stl file and uploaded it to Cura. I used the default settings for ender 3 and then proceeded to slice the file. I used the x ray mode to ensure the clip was one body and I also used layer mode to preview how the nozzle would move. Everything looks fine so I loaded and saved the file to my sd card, I proceeded to print the file on my printer but the nozzle doesn't move as shown on layer view. I uploaded and saved a similar design from Thingiverse with the same results print keeps cutting off on one end. In layer view the clip gets outlined then filled in  but for me, it gets outlined to an extent then starts filling in like on the left side print on actual results.
**Screenshot(s)**

**Actual results**
[image]

**Expected results**
[image]

**Project file**

[image]

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

closed 2020-05-06 by aralas77

Ghostkeeper, 2020-05-07:
Did you accidentally close this, or did you find a solution?

This seems like maybe a question of making sure your bed is level, rather than anything that Cura can do for you.

aralas77, 2020-05-17:
Sorry for the late response, I figured out the problem. I forgot to add the new dimensions on firmware so printer was limited by old dimensions. Thanks for the help!