Ultimaker/Cura #7688: cura brim error

Richi956, 2020-05-06:
**Application version**
(The version of the application this issue occurs with.)
V 4.6.1
**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
Windows 10 Pro 64-bit
Nividia GTX 1060

**Printer**
(Which printer was selected in Cura?)
Sovol Sv01
**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)
brim on
**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)
brim not on all sufaces
[image]

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)
crua creats brim but not on all sufaces. print fail.

mahtDFR, 2020-05-06:
It's possible your model isn't completely flat on the build plate. Check that first.
You can lay the model flat by selecting the "Rotate" tool and choosing "Lay flat" in the flyout menu.

Hope this helps.

Richi956, 2020-05-06:
Thanks, now it works.
[image]
[image]

closed 2020-05-06 by Ghostkeeper