Ultimaker/Cura #7689: USB Printer ?

Tchicken95, 2020-05-06:
Good morning all,
It is impossible to find the plugin 'USB Printer' is this normal ?
Thank's for your help, Thcicken.

mahtDFR, 2020-05-06:
Can you please fill in the issue template that you deleted to type this, otherwise we have to try and guess what you mean

closed 2020-05-06 by Tchicken95

Tchicken95, 2020-05-06:
My printer is a DaVinci 2.0a Duo flashed with Repetier, I am in CURA version 4.6.1

mahtDFR, 2020-05-06:
D'accord, remplir le formulaire, s'il vous plaît:

**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

Tchicken95, 2020-05-06:
Application version
CURA 4.6.1
Repetier FirmWare 0.92

Platform
Windows 10 1909 
GPU Intel Graphiq 4000

Printer
Custom FFF Printer
[image]
[image]

Actual results
[image]

Expected results
Unable to send to printer

Project file
[PieceCouverclePoubelle.zip](https://github.com/Ultimaker/Cura/files/4585963/PieceCouverclePoubelle.zip)

Log file
(See https://github.com/Ultimaker/Cura#logging-issues
[cura.log](https://github.com/Ultimaker/Cura/files/4585976/cura.log)

Additional information
Cura Config :
[cura.zip](https://github.com/Ultimaker/Cura/files/4585978/cura.zip)

Tchicken95, 2020-05-06:
Arduino Due on port COM7 :

[image]

Ghostkeeper, 2020-05-08:
Thank you. This entry from the log makes it clear I think:

```
2020-05-06 00:55:01,667 - DEBUG - [Thread-1] USBPrinting.AutoDetectBaudJob.run [50]: Checking COM7 if baud rate 115200 works. Retry nr: 1. Wait timeout: 15
2020-05-06 00:55:01,679 - WARNING - [Thread-1] UM.Logger.logException [106]: Exception: Unable to create serial
2020-05-06 00:55:01,685 - WARNING - [Thread-1] UM.Logger.logException [110]: Traceback (most recent call last):
2020-05-06 00:55:01,692 - WARNING - [Thread-1] UM.Logger.logException [110]:   File "C:\Program Files\Ultimaker Cura 4.6\plugins\USBPrinting\AutoDetectBaudJob.py", line 54, in run
2020-05-06 00:55:01,697 - WARNING - [Thread-1] UM.Logger.logException [110]:     serial = Serial(str(self._serial_port), baud_rate, timeout = read_timeout, writeTimeout = write_timeout)
2020-05-06 00:55:01,703 - WARNING - [Thread-1] UM.Logger.logException [110]:   File "serial\serialwin32.py", line 31, in __init__
2020-05-06 00:55:01,709 - WARNING - [Thread-1] UM.Logger.logException [110]:   File "serial\serialutil.py", line 240, in __init__
2020-05-06 00:55:01,715 - WARNING - [Thread-1] UM.Logger.logException [110]:   File "serial\serialwin32.py", line 62, in open
2020-05-06 00:55:01,720 - WARNING - [Thread-1] UM.Logger.logException [110]: serial.serialutil.SerialException: could not open port 'COM7': PermissionError(13, 'Accès refusé.', None, 5)
2020-05-06 00:55:01,721 - DEBUG - [Thread-1] USBPrinting.AutoDetectBaudJob.run [50]: Checking COM7 if baud rate 250000 works. Retry nr: 1. Wait timeout: 15
```

Access to COM7 is "refusé"d :grin: 

Probably something else is connected on that serial port. Do you have two Cura programs running? Or maybe something like the Arduino IDE?
Feeble attempt at French: Je pense que quelque chose d'autre est connecté à ce port COM. Avez-vous exécuté le Cura deux fois en même temps? Ou peut-être l'IDE de Arduino?

Tchicken95, 2020-05-08:
Thank you and sorry, Tchicken.