Ultimaker/Cura #7691: (Crash) Incomplete transfer of settings from the old version to the new version (from cura 4.5 to cura 4.6.1)

Goodfeat, 2020-05-06:
found a small error while updating the program.
the program does not transfer files correctly

I have a separate printer, its settings are based on an ultimaker.
[image]

I decided to update the program and was faced with the fact that the program crashes when switching to another printer

the problem was found right away, so no reports needed

it turns out when updating the program does not change the version settings to the current

[image]

I needed to go into the settings options and fix the setting_version parameter manually.

from setting_version = 11 
to  setting_version = 13

[image]

After that everything worked

If I’m not mistaken, usually when transferring the settings, the program itself should fix the settings version. for some reason this does not happen with this folder

nallath, 2020-05-06:
This is correct. We don't have an upgrade for the variants. The upgrade scripts generally only update profiles that can be created through the interface.

Ghostkeeper, 2020-05-11:
Yes, I agree. This is outside of the scope of the version upgrade at the moment, in order to keep things a bit more simple for us. We don't want to be testing the version upgrade with manually created profiles.

closed 2020-05-11 by Ghostkeeper