Ultimaker/Cura #7692: Have "Preview" , "GCode" and "Post-pro" tools indiacte the same layer number.

jpmath54, 2020-05-06:
Hi,

I've just struggled to slow down my last layer on my current project.

For sake of clarity and simplicity, this little change would avoid troubles to users.
"Preview" tab should start with layer "0", the one that will provide adhesion on plate.
This change wouldn't impact G-CODE generation.

That's it.

Regards,
JPMT

widget-, 2020-05-06:
Just spent a couple hours on this. I'm trying to do some rapid prints and wanted the printer to start cooling when it hit the last layer (layer 4 in the preview.) Apparently I need to set the ChangeAtZ script to cool the bed at layer "2" to do that.

Ghostkeeper, 2020-05-11:
We have an open bug with the [Pause At Height script](https://github.com/Ultimaker/Cura/issues/5085) regarding the layer that it chooses to pause at. It's sometimes inconsistent with layer view and sometimes not, depending on which height you choose to pause at. We've also discovered that this holds for ChangeAtZ too, though it wasn't reported here on Github.

It is our intention to start counting at 1 like normal people, not at 0 like programmers. The post-processing scripts should be in sync with the layer view layer numbers though.

jpmath54, 2020-05-13:
I think programmers are the only "normal" people.
Do whatever suits you, but please consider that the issue is not only with the GUI, but also concerns the GCODE that will be produced.
By now, the first layer is always designated as : `;LAYER:0`

Ghostkeeper, 2020-05-14:
Yes, the g-code on the other hand is considered to be read by programmers :smile:

The g-code is being read and parsed by a lot of things. I honestly think that staying consistent with our historic output is more important than staying consistent with what is being shown to the user. Do you think we should change it?

fieldOfView, 2020-05-14:
>  I honestly think that staying consistent with our historic output is more important than staying consistent with what is being shown to the user.

I don't.

> Do you think we should change it?

Yes.

mahtDFR, 2020-05-19:
We discussed this and don't feel comfortable changing it

closed 2020-05-19 by mahtDFR

fieldOfView, 2020-05-19:
The longer you keep the inconsistency, the more uncomfortable it will be to fix it.

Ghostkeeper, 2020-05-19:
There are two major reasons to not change this:
* The use case of reading g-code files is not meant to be something that humans need to do. Yes, you can open up the UFP file as a zip and read the g-code, or open the g-code directly in the case of other printers, but actually understanding the file (e.g. to verify where the pause from a Pause at Height post-processing script) is considered to be an expert-level action. It shouldn't be necessary for humans to do that. It should be necessary for computer programs, but they tend to work with 0-indexing anyway, aside from Lua. So the use case is very small.
* If we change this, all programs that interpret these layer numbers (including Cura itself) will need to support both formats. That's the "uncomfortable" we are talking about. It leads not only to bugs in Cura, but also some [OctoPrint plug-ins](https://plugins.octoprint.org/plugins/DisplayLayerProgress/), post-processing scripts people may have themselves (though not in Cura as far as I know) and even the Ultimaker 2 firmware. We're not likely going to be publishing new UM2 firmware, so that pretty much defeats the feature request on its own.

jpmath54, 2020-05-19:
I think this is your developer choice, and that developers always make the right decisions.

Anyway, at least, please consider to: 
- choose wether **ALL** Cura GCODE post-pro tools should be consistent with "gcode numbering" or "GUI numbering". As they are dedicated to advanced users, and post process the GCODE, I think they should fit to GCODE numbering for sake of clarity, but who am I to decide?
- document this behavior clearly in the post-pro section.

widget-, 2020-05-19:
I think documentation within the UI about which layers and at what point within the layers the gcode is inserted is still required, even if it's an "expert" feature. As of now the plugin shows nothing.

Ghostkeeper, 2020-05-22:
The post-processing plug-ins are meant to make their human-visible layer numbers 1-indexed indeed. For that issue, see https://github.com/Ultimaker/Cura/issues/5085.