Ultimaker/Cura #7694: clean top layer for reflective metal filaments

dennik, 2020-05-06:
**Is your feature request related to a problem? Please describe.**
Ive tried to print with silver PLA and noticed that when the top layer has a hole or several holes on it, the path gets interrupted, the printhead jumps to a different location, and by the time it fills up all the gaps, unpleasing patches appear cause the plastic reflects light in a different way from patch to patch. 
[image]
The reason this is happening is unknown to me but I imagine it happens with most reflective metal filaments.

**Describe the solution you'd like**
(A clear and concise description of what you want to happen. If possible, describe why you think this is a good solution.)

I would hope for an option to close top layers removing any vertical through holes to avoid the print path getting interrupted. That way all flat areas are printed evenly and cleanly before the next layer starts.

**Describe alternatives you've considered**
I have already simulated this solution by separating the letters from the flat surface below them by 0.02mm. Enough so that the slicer closes that through hole under each letter, and starts the letter on the next layer. Adhesion and structural integrity is probably even better than with just fill in walls underneath. The gap needs to increase if the layer height increases so that the slicer doesn't re generate that hole.
[image]
The result is a much cleaner print with the patches gone.

**Affected users and/or printers**
(Who do you think will benefit from this? Is everyone going to benefit from these changes? Or specific kinds of users?)
Anyone who prints with reflective materials could benefit from this. I imagine most people would love to have clean looking top layers without those patches showing up. I imagine this problem might not show up with matte finish materials, or flat surfaces without through holes.
I've only tested this on a UM3 printer.

**Additional context**
(Add any other context or screenshots about the feature request here.)

64Florian, 2020-05-07:
Hello,
You have this feature / workaround using top / bottom layer to 0 and  infill 100%:
[image]
[image]

I attached my file (I sliced @ 0.1 mm/layer)
[TestJoy.stl.zip](https://github.com/Ultimaker/Cura/files/4590058/TestJoy.stl.zip)

smartavionics, 2020-05-07:
Even simpler, just use a larger value of top skin expand distance....

[image]

mahtDFR, 2020-05-07:
Good suggestions above. 'Enable Ironing' might also improve your results.
It seems there are sufficient solutions already so I don't think we're going to do more with this.

64Florian, 2020-05-07:
> Even simpler, just use a larger value of top skin expand distance....

Doesn't work for me ... (@smartavionics did you try to "play" the layer ? )
[image]

smartavionics, 2020-05-07:
You probably also need to set top surface skin layers to 0.

64Florian, 2020-05-07:
> You probably also need to set top surface skin layers to 0.
As I can see in my tests, if skin layers = 0 I have to set infill to 100% 
with this settings I'm not able to see any influence for skin expand distance but I could have miss something ...

@mahtDFR  I have try to see for ironing, the finish could be better but it could have the same issue too because: this issue is linked to the way the head add filament and move on the surface 
-> To have all particulate in the same "way" you have to apply exactly the same travel 
unfortunately it's not possible for zigzag mode .. 
[image]
[image]

it could give a good result for concentric mode (if the last layer doesn't break definitively the chances to made a good finish (Tests has to be done))
[image]

dennik, 2020-05-07:
> 
> 
> Good suggestions above. 'Enable Ironing' might also improve your results.
> It seems there are sufficient solutions already so I don't think we're going to do more with this.

> 
> 
> Even simpler, just use a larger value of top skin expand distance....
> 
> [image]

Yes I can confirm that top skin expand distance is doing the same trick in a different way.

dennik, 2020-05-07:
> 
> 
> Good suggestions above. 'Enable Ironing' might also improve your results.
> It seems there are sufficient solutions already so I don't think we're going to do more with this.

Ironing alone won't do the trick. It will just smear the same reflectivity difference in other directions.
[image]
it does enhance the quality though on a clean top layer.

64Florian, 2020-05-07:
@dennik did you try my settings ? 0 top and infill 100% lines infill:

> [image]

as I can see in "play" mode in preview, it should be OK for full clean layer

dennik, 2020-05-07:
> 
> 
> @dennik did you try my settings ? 0 top and infill 100% lines infill:
> 
> > [image]
> 
> as I can see in "play" mode in preview, it should be OK for full clean layer
Yes, the 100% infil definitely closes all holes although it adds to the 3D print time and also filament consumption. I didn't see any difference with the top/bottom thickness setting though.

64Florian, 2020-05-07:
"Bottom" is overkill .. but with my test if you have 1 top layer you're back to the issue (as I can see when you "play" the layer)
If you want to preserve your print time / filament consumption, you have to add a second object to "modify settings on overlaps" as said before:

> P.S. if you want an empty product or a 20% infill for the bottom, you can add another shape and use "modify settings on overlaps" or the other options to customize the bottom of the product
> [image]

Have you ever use this feature ? (_"modify settings on overlaps"_)

dennik, 2020-05-07:
> "Bottom" is overkill .. but with my test if you have 1 top layer you're back to the issue (as I can see when you "play" the layer)
> If you want to preserve your print time / filament consumption, you [...]

That's a great feature to use. Had no idea about the functionality of that button. For now I think it is probably best to use the "top skin expand distance" as  mahtDFR suggested. Although it is a hard setting to find unless you know exactly what you are looking for.

closed 2020-05-11 by Ghostkeeper