Ultimaker/Cura #7695: [3.3.1]setting "Builtplate temperature" to zero leaves the "builtplate temperature initial layer" setting active for th whole print

mcmeyer89, 2020-05-06:
**Application version**
3.3.1

**Platform**
Windows 10

**Printer**
Anet A8

**Reproduction steps**
1. Set "buildplate tamperature" to zero while initial "buildplate tamperature initial layer" is set to 110 and slice.
2. Gcode does not contain M140 to change tamperature after first layer
3. Verified by only changing "buildplate tamperature" to 1
4. G-code does contain M140 S1

I was using this settings to get good layer buildbed adhesion but not heating the bed and build chamber during the hole print
Reasons: 
1.energy saving
2. Eas of mind to have the bed switched off for long prints.
3. In this context it could also be considered a safety issue when the printer is left unattended expecting the bed to be switched off after first layer.

**Screenshot(s)**
N/A

**Actual results**
No M140 in G-code

**Expected results**
"M140 S0" in G-code after first layer

**Project file**
[Builtblate_temperature_bug.zip](https://github.com/Ultimaker/Cura/files/4588712/Builtblate_temperature_bug.zip)

mahtDFR, 2020-05-07:
Can you check your start gcode in machine settings? Maybe there's a command in there that's been missed.

Ghostkeeper, 2020-05-11:
I could swear we've seen this issue reported before, but I can't find it anywhere.

But yes this also reproduces in current master. No `M140` or `M190` to change the temperature.

louiswouters, 2020-06-08:
I wrote a postprocessing plugin to change the temperatures of the bed and/or nozzle during print after a certain layer:
https://github.com/louiswouters/3D-Printing/blob/master/scripts/ChangeTemperatureDuringPrint.py