Ultimaker/Cura #7697: Issues Slicing .8 mm walls when using .8 mm nozzle

benderprod, 2020-05-06:
**Application version**
4.6.0

**Platform**
Windows 10

**Printer**
Ender 5

**Reproduction steps**
1. Generated a model in Fusion 360 with walls that are .8mm thick.  Rectangular or Circular
2. Slice in Cura with .8mm nozzle selected.

**Screenshot(s)**
Slicing .8mm walls with .8mm nozzle
[image]

Slicing 1.6mm walls with .8mm nozzle
[image]

**Actual results**
Wall are incomplete, leaving opened gaps.

**Expected results**
Each layer should simply be a single wall circle.

**Project file**
[Outside Speaker Light - Outside Spiral (0.8 walls).3mf.zip](https://github.com/Ultimaker/Cura/files/4589774/Outside.Speaker.Light.-.Outside.Spiral.0.8.walls.3mf.zip)

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/4589780/cura.log)

64Florian, 2020-05-07:
I saw this "issue" too ... (for 0.4 mm nozzle)
I use "spiralize outer contour as workaround 
[image]
"
but in your case, you may have to use another object with "modify object for overlaps" for your 4 handles (they look like thicker than the body)

Ghostkeeper, 2020-05-11:
That's because your model has staggered vertices for the inside and outside of the model:

[image]

This means that in some places the wall is thinner than 0.8mm and in some places the wall is wider than 0.8mm. The wall only gets drawn if it's wider than one line width.

Duplicate of https://github.com/Ultimaker/Cura/issues/7687.

closed 2020-05-11 by Ghostkeeper