Ultimaker/Cura #7698: Direct access to "Slice automatically"

64Florian, 2020-05-06:
**Is your feature request related to a problem? Please describe.**
I'm always frustrated when I change a setting with a big STL file and the opposite when it's a small one and the setting is disable

**Describe the solution you'd like**
Please add a checkbox:
[image]

**Affected users and/or printers**
Everyone going to benefit from these changes. -> time saver than settings -> pref-> check -> close
and better workflow for beginners and basics users how don't see this feature and have to wait or click as fast as the can on cancel button when they fine tune settings of they profiles (in preview mode)

yet-another-average-joe, 2020-05-07:
It seems you just need the "Slicing Toggle by fieldOfView" plugin. Did you give it a try ? It does exactly what you're asking for, with a "run/pause" toggle graphic button :

https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/PauseBackendPlugin

mahtDFR, 2020-05-07:
The above is correct, this feature already exists via a plugin. Will close this now.

closed 2020-05-07 by mahtDFR

64Florian, 2020-05-07:
Thanks a lot !
That's exactly what I wanted