Ultimaker/Cura #7701: Ironing speed for 1st (slowdown) layer

maukcc, 2020-05-07:
**Application version**
4.6 PPA

**Platform**
Linux

**Printer**
HMS434

**Reproduction steps**
1. ironing the 1st layer
2. 1st layer is a "slowdown" layer

**Actual results**
ironing speed is reduced due to slowdown layer 

**Expected results**
normal speed for ironing

**Additional information**
Is this bug or feature?

mahtDFR, 2020-05-07:
Are you able to provide some more context here? Do you have a project file?

Ghostkeeper, 2020-05-11:
I think what you're mentioning is intended, not a bug. The slower speed at the initial layer is meant to slow down all movements aside from the skirt/brim which have their own speed setting that only applies to the first layer anyway.

After all, if you were to print the ironing using the normal fast speed you could tear off the print from the build plate anyway.

maukcc, 2020-05-11:
OK.

closed 2020-05-11 by maukcc