Ultimaker/Cura #7702: Infill Speed bug

SJC28, 2020-05-07:
Hi.

I have the infill speed selected in settings, but it does not appear under Speed when I want to modify it. If I select any other speed setting it works perfectly.

[image]

smartavionics, 2020-05-07:
Perhaps infill density is 0?

SJC28, 2020-05-07:
The Infill it's at 15%

64Florian, 2020-05-07:
Have you try:
 - to search for "infill" in the search settings area ? 
 - to disable the visibility, close cura, reopen, and re-enable the visibility ? 
 - to use another profile

SJC28, 2020-05-07:
Yes. I have tried that and nothing. Also tried updating Cura and still
nothing.

The only thing that helped was uninstalling Cura and every profile I had
and then install it again and create all the profiles again.

Ghostkeeper, 2020-05-11:
Could you provide a project file that reproduces the issue? It's most likely one of your setting overrides that makes it disappear.

closed 2020-06-19 by no-response[bot]