Ultimaker/Cura #7703: [4.6.1] Scaling Y Z mixup

Jvejvad, 2020-05-07:
**Application version**
4.6.1
(The version of the application this issue occurs with.)

**Platform**
Windows 10 (64)
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
Anet A8 (custom setup)
(Which printer was selected in Cura?)

**Reproduction steps**
1. load any STL
2. Chaning Z hight when not using uniform scaling will change hight, but change Y value number in the scaling box. (and vice versa - Y change number is displayed as Z)

**Screenshot(s)**
before:
[image]

after
[image]

(Image showing the problem, perhaps before/after images.) 

**Actual results**
The stl will change but it's confusing and wrong.
(What happens after the above steps have been followed.)

**Expected results**
Normaly the value change according to the axis changed.
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

mahtDFR, 2020-05-07:
I'm afraid I couldn't reproduce this over here. It works as expected for me. Maybe you can share a project file?

> Anet A8 (custom setup)

Does that mean you're not using the pre-shipped profile? Can you give me some more details, please?

Jvejvad, 2020-05-07:
Hi.
I will try to reproducere on another predefined profile and get back to you with more info.

Thanks.

Mvh Jacob Varming Vejvad

dennik, 2020-05-07:
I can confirm that there is something iffy going on with the non uniform scale and it happens for many versions now. In my case:
I select a model, change x to 50%
I hit enter, both x and z change to 50%
Trying to correct that, all hell breaks loose.
The problem is with the numeric value input, not the direct mouse drag on the scale axis.

dennik, 2020-05-07:
I think I found what the issue is. You can replicate it on your own if you rotate the model first before scaling. Then try to change a numeric value by hand. Looks like prior rotations confuse non uniform scale tool.

dennik, 2020-05-07:
After this gets fixed, I suggest both global and local axis transformation for that tool. Just add one more check box on the panel for local transformation and preferably have that checked by default.

mahtDFR, 2020-05-08:
Ah yes, after rotating the model and trying to scale everything to 100%, I also saw some weird behaviour. Will discuss

Ghostkeeper, 2020-05-08:
Duplicate of https://github.com/Ultimaker/Cura/issues/4136.

closed 2020-05-08 by Ghostkeeper