Ultimaker/Cura #7706: Brim enhancement

Sophist-UK, 2020-05-07:
**Is your feature request related to a problem? Please describe.**
Make it easier to break off the brim cleanly.

**Describe the solution you'd like**

Assuming that there are at least two layers of empty space above the second-most inner brim line, print an extra layer on that line to make it more likely that the brim will break off at the innermost brim line.

**Describe alternatives you've considered**
This could, I believe, be provided as a post-processing script.

**Affected users and/or printers**
All.

mahtDFR, 2020-05-07:
Have you used the brim distance setting that smartavionics contributed? It was introduced in 4.5. Using a 0.4 nozzle with a brim distance value of 0.4 as well works really well for me.

Sophist-UK, 2020-05-07:
Yes - but I sometimes find that the brim tears close to, but not next to, the object. I thought this might help reduce tearing of the brim near the object.

mahtDFR, 2020-05-07:
Couldn't you just fine-tune the value until you get the performance you need?
Maybe you can visualise your proposal for me, I'm not sure I understand completely.

Ghostkeeper, 2020-05-11:
I think he's talking about this problem where there are like 2 lines left of the brim:

[image]

I pretty much always use a knife and then it's not an issue, but if you're breaking it off by hand and you've just cut your nails it can be annoying when it's breaking off at 1 line away from the wall.
This proposal is to prevent that from happening, by making the brim stronger except where it needs to break off.

I think it's a pretty minor problem though. And one that could be solved by making sure that your initial layer flow is good and maybe some brim distance.