Ultimaker/Cura #7709: Cannot see any default profiles

aguilaair, 2020-05-07:
**Application version**
(The version of the application this issue occurs with.)
4.6.1

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
Windows 10 Pro Insider, running an 17 6th gen and a GTX1080

**Printer**
(Which printer was selected in Cura?)
Creality Ender 3

**Reproduction steps**
1. Open Cura and add a new printer
2. Click on the profiles tab to check the available ones.

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
[image]

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)
Being able to select a preset

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)
https://pastebin.com/raw/RFJdUYt3

**Additional information**
(Extra information relevant to the issue.)
I have tried doing the following:
Resetting Cura
Reinstalling Cura
Removing and adding the printer
Checked material is set on Generic PLA

64Florian, 2020-05-07:
Have you try to remove all profile on your system ?
 - Help -> show configuration folder -> File explorer should open and located in a 4.6 folder ...
 - close CURA
 - rename "cura" parent folder (of 4.6) 
 - restart Cura -> you should have the "welcome screen"

64Florian, 2020-05-07:
After looking in the logs .. 
For technical Ultimaker teams, according the log it seem to be another "setting_version': '11'" issue 
-> not upgraded profile from an older version

nallath, 2020-05-08:
I don't think it's exactly that issue. It seems that a bunch of artillery sidewinder files were sideloaded. Those should obviously not affect the creality printer in any way. 

If you haven't deleted your configuration folder, could you zip it and send it over? I could have a look to see what is going on.

Ghostkeeper, 2020-05-11:
Yup, see also this related issue that came in from the Artillery Sidewinder users: https://github.com/Ultimaker/Cura/issues/7636

The problem is that someone created Artillery Sidewinder profiles but gave them the same filenames as Creality profiles. Cura will then pick the first one it sees and ignore the one with a duplicate ID. Those profiles are faulty.

Another problem maybe is that the built-in Creality quality profiles don't have the word "creality" in their file names. That makes them vulnerable to this mix-up.

closed 2020-05-11 by Ghostkeeper