Ultimaker/Cura #7711: [4.6.1] Jump in tree support diameter results in failed print

Bflat703, 2020-05-08:
**Application version**
4.6.1
**Platform**
MacOS 10.13.6 on Mac Pro 3.2 GHz Quad-Core Intel Xeon w/ ATI Radeon HD 5770 1024 MB
**Printer**
Creality CR-10 

**Reproduction steps**
1. Slice with 'Generate Supports' OFF
2. and 'Tree Supports' ON

**Screenshot(s)**
[image]

**Actual results**
8 filament diameter jump in tree support diameter in 1 layer.

**Expected results**
Tree support should increase in diameter in a normal way.

**Project file**

[CCR10_T1_Housing_Support_Fail.txt](https://github.com/Ultimaker/Cura/files/4596704/CCR10_T1_Housing_Support_Fail.txt)

**Log file**
2020-05-07 18:34:37,856 - DEBUG - [MainThread] UM.Controller.setActiveStage [180]: Setting active stage to PreviewStage
2020-05-07 18:34:37,870 - DEBUG - [MainThread] UM.Controller.setActiveView [108]: Setting active view to SimulationView
2020-05-07 18:41:29,467 - DEBUG - [MainThread] cura.AutoSave._onTimeout [60]: Autosaving preferences, instances and profiles
2020-05-07 18:41:32,588 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._rescheduleCheck [135]: New error check scheduled.
2020-05-07 18:41:33,430 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [187]: Error check finished, result = False, time = 0.8s
2020-05-07 18:41:41,468 - DEBUG - [MainThread] cura.AutoSave._onTimeout [60]: Autosaving preferences, instances and profiles
2020-05-07 18:41:52,047 - DEBUG - [MainThread] UM.Controller.setActiveStage [180]: Setting active stage to PrepareStage
2020-05-07 18:41:52,054 - DEBUG - [MainThread] UM.Controller.setActiveView [108]: Setting active view to SolidView
2020-05-07 18:41:52,120 - DEBUG - [MainThread] cura.Machines.Models.NozzleModel._update [28]: Updating NozzleModel.
2020-05-07 18:41:52,202 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///Applications/Ultimaker Cura.app/Contents/MacOS/resources/qml/ActionButton.qml:101: ReferenceError: contentWidth is not defined
2020-05-07 18:41:52,302 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///Applications/Ultimaker Cura.app/Contents/MacOS/resources/qml/ActionButton.qml:101: ReferenceError: contentWidth is not defined
2020-05-07 18:41:52,342 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/Applications/Ultimaker Cura.app/Contents/MacOS/resources/shaders/xray.shader]...
2020-05-07 18:41:57,679 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [237]: Starting to slice...
2020-05-07 18:41:57,680 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [255]: Going to slice build plate [0]!
2020-05-07 18:41:57,693 - DEBUG - [MainThread] UM.Backend.Backend._createSocket [227]: Previous socket existed. Closing that first.
2020-05-07 18:41:57,704 - WARNING - [Thread-3] UM.Decorators.deprecated_function [23]: <function GlobalStack.extruders at 0x608000f02ad8> is deprecated (since 4.4): Please use extruderList instead.
2020-05-07 18:41:57,777 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [184]: Socket state changed to Listening
2020-05-07 18:41:58,147 - INFO - [MainThread] UM.Backend.Backend.startEngine [86]: Started engine process: /Applications/Ultimaker Cura.app/Contents/MacOS/CuraEngine
2020-05-07 18:41:58,193 - DEBUG - [MainThread] UM.Backend.Backend._backendLog [106]: [Backend] Calling engine with: ['/Applications/Ultimaker Cura.app/Contents/MacOS/CuraEngine', 'connect', '127.0.0.1:49674', '']
2020-05-07 18:41:58,317 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [106]: [Backend] 
2020-05-07 18:41:58,326 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [106]: [Backend] Cura_SteamEngine version 4.6.1
2020-05-07 18:41:58,335 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [188]: Socket state changed to Connected
2020-05-07 18:41:58,358 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [106]: [Backend] Copyright (C) 2019 Ultimaker
2020-05-07 18:41:58,361 - DEBUG - [MainThread] UM.Backend.Backend._onSocketStateChanged [177]: Backend connected on port 49674
2020-05-07 18:41:58,370 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [106]: [Backend] 
2020-05-07 18:41:58,398 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [106]: [Backend] This program is free software: you can redistribute it and/or modify
2020-05-07 18:41:58,419 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [106]: [Backend] it under the terms of the GNU Affero General Public License as published by
2020-05-07 18:41:58,428 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [106]: [Backend] the Free Software Foundation, either version 3 of the License, or
2020-05-07 18:41:58,433 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [106]: [Backend] (at your option) any later version.
2020-05-07 18:41:58,437 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [106]: [Backend] 
2020-05-07 18:41:58,441 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [106]: [Backend] This program is distributed in the hope that it will be useful,
2020-05-07 18:41:58,446 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [106]: [Backend] but WITHOUT ANY WARRANTY; without even the implied warranty of
2020-05-07 18:41:58,451 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [106]: [Backend] MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
2020-05-07 18:41:58,456 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [106]: [Backend] GNU Affero General Public License for more details.
2020-05-07 18:41:58,462 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [106]: [Backend] 
2020-05-07 18:41:58,467 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [106]: [Backend] You should have received a copy of the GNU Affero General Public License
2020-05-07 18:41:58,471 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [106]: [Backend] along with this program.  If not, see <http://www.gnu.org/licenses/>.
2020-05-07 18:41:58,685 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onStartSliceCompleted [444]: Sending slice message took 1.0056369304656982 seconds
2020-05-07 18:41:58,808 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [106]: [Backend] [WARNING]  -s magic_mesh_surface_mode="normal" -s machine_max_acceleration_y="500" -s material_bed_temperature="60" -s infill_sparse_thickness="0.2" -s speed_travel="150.0" -s date="07-05-2020" -s infill_support_angle="40" -s mesh_position_y="0" -s infill_multiplier="1" -s raft_surface_speed="25.0" -s z_seam_y="300" -s speed_support_interface="25.0" -s bridge_skin_support_threshold="50" -s speed_wall_x="25.0" -s infill_material_flow="100" -s support_extruder_nr="0" -s wireframe_flat_delay="0.1" -s extruders_enabled_count="1" -s support_brim_width="4" -s retraction_min_travel="1.5" -s top_skin_preshrink="0.8" -s raft_interface_fan_speed="0" -s magic_spiralize="False" -s support_brim_enable="True" -s jerk_ironing="8" -s magic_fuzzy_skin_point_density="1.25" -s support_pattern="zigzag" -s machine_shape="rectangular" -s support_interface_density="33.333" -s sub_div_rad_add="0.4" -s machine_heat_zone_length="16" -s initial_extruder_nr="0" -s minimum_polygon_circumference="1.0" -s material_bed_temp_wait="True" -s prime_tower_wipe_enabled="True" -s infill_randomize_start_location="False" -s anti_overhang_mesh="False" -s speed_travel_layer_0="100.0" -s material_end_of_filament_purge_speed="0.5" -s infill="0" -s infill_wipe_dist="0.0" -s alternate_carve_order="True" -s speed_roofing="25.0" -s machine_max_acceleration_e="5000" -s jerk_enabled="False" -s small_feature_speed_factor_0="50" -s material_guid="" -s roofing_angles="[ ]" -s outer_inset_first="False" -s jerk_wall="8" -s retraction_retract_speed="25" -s fill_perimeter_gaps="everywhere" -s support_roof_height="0.8" -s optimize_wall_printing_order="True" -s travel="0" -s ooze_shield_dist="2" -s infill_offset_y="0" -s material_print_temperature="210" -s max_skin_angle_for_expansion="90" -s prime_blob_enable="False" -s speed_layer_0="20.0" -s material_no_load_move_factor="0.940860215" -s wireframe_up_half_speed="0.3" -s roofing_pattern="lines" -s machine_acceleration="500" -s support_bottom_extruder_nr="0" -s support_use_towers="False" -s speed_print="50.0" -s retraction_count_max="100" -s machine_steps_per_mm_x="50" -s speed_topbottom="25.0" -s raft_smoothing="5" -s cool_min_layer_time_fan_speed_max="10" -s machine_nozzle_heat_up_speed="2.0" -s materia [... cut ...]
2020-05-07 19:29:11,127 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._processNextRequestsInQueue [317]: No more requests to process, stop
2020-05-07 19:29:11,127 - INFO - [MainThread] SliceInfoPlugin.SliceInfo._onRequestFinished [275]: SliceInfo sent successfully
2020-05-07 19:29:40,614 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///Applications/Ultimaker Cura.app/Contents/MacOS/resources/qml/RoundedRectangle.qml:68: TypeError: Cannot read property of null

**Additional information**
Turning on both 'Generate Supports' and 'Tree Supports' fixes the problem but generates

Ghostkeeper, 2020-05-11:
Your project file is not a project file, but a g-code file.

We've seen this problem before in the past, but I thought it was fixed. Apparently not. Could you still upload an actual project file? Like, just press Cmd+S and upload that file.

Bflat703, 2020-05-12:
I tried this on another machine and was unable to reproduce the problem. This was a completely default installation of Cura so I started playing with the settings. What I found was that having certain support settings enabled in preferences brought the problem back even though all of them except "Generate Supports" and "Touching Buildplate" are left at their default values. I found that turning on "Generate Supports" does not really fix the problem, it just allows the print to complete because the straight support 'supports' the tree support. After I changed the preferences the problem is completely repeatable. This is a picture of the settings:
[image]

Project File:

CMD+S allows me to save a copy of this website on my machine.  STL is not a supported file type. I'll try emailing it to you.

Bflat703, 2020-05-12:
CMD+S just lets me save a copy of the GitHub website on my local machine.  STL is not a supported file type for uploads.  This is the only way I can think of to forward a project file.

mahtDFR, 2020-05-12:
Please save a project file (File -> Save...), zip up the .3mf file and attach to this issue so I can try to reproduce it with the same settings as you.

Ghostkeeper, 2020-05-12:
> CMD+S allows me to save a copy of this website on my machine.

That's when you press CMD+S while browsing this website. What I meant was in Cura.

> STL is not a supported file type for uploads.

The STL probably won't allow me to reproduce the issue. We need the project file. To upload it on Github we'd need it zipped.

> This is the only way I can think of to forward a project file.

You are sending an e-mail to github.com. The website is kind enough to post your e-mail on the issue thread on Github but it doesn't put attachments there so it won't work that way, sorry.

Bflat703, 2020-05-13:
Ok, now I get it.  Please forgive me, I've only been using Cura since the beginning of March.
Here's the file:
[CCR10_Housing_with Hinge.3mf.zip](https://github.com/Ultimaker/Cura/files/4619384/CCR10_Housing_with.Hinge.3mf.zip)

Ghostkeeper, 2020-05-13:
Okay, thanks!

Hrrrrr, the problem indeed reproduces in Cura 4.6.1, but not in 4.7 any more... Probably due to vertices SLIGHTLY shifting around due to the new polygon simplification changes, I guess. We didn't change anything in this algorithm since 4.6 as far as I know. At least we can trace the problem as it happens now. I'll see what we can do.

Bflat703, 2020-06-24:
CMD+S just lets me save a copy of the GitHub websight on my local machine.  STL is not a supported file type.  This is the only way I can think of to forward a copy of the file.

Ghostkeeper, 2020-07-01:
CMD+S produces a project file which is an adjustment of a 3MF file.

Github doesn't allow STL or 3MF to be uploaded so you must put them in a .zip archive. That is allowed. They probably did this to stimulate people to compress their files before uploading them.

Waldkaeuzchen, 2020-09-07:
Hello,

I have the same problem with a project. I am using version 4.7 and was able to repeat the problem with different support settings. As soon as the model is moved slightly the error does not occur anymore and the support is generated normally. In the appendix you can find the corresponding project
[image]

[UM3_lwing_laat.zip](https://github.com/Ultimaker/Cura/files/5182325/UM3_lwing_laat.zip)

Ghostkeeper, 2020-09-08:
Yup, with that project the bug reproduces for me as well.

closed 2020-09-10 by Ghostkeeper

Ghostkeeper, 2020-09-11:
Should be fixed in 4.8.