Ultimaker/Cura #7712: If I make a patch to enable an option for the mouse to behave the same as Fusion 360, will you merge it?

gitcnd, 2020-05-08:
**Is your feature request related to a problem? Please describe.**

Default Fusion 360 mouse controls and Cura ones are different, creating mass confusion

**Describe the solution you'd like**

I can do a patch to fix this (a non-default option that users can enable), but, before I put in the effort, is there anyone in control of this project who is "highly likely" to merge this in?

I do not want to waste mine nor your time...

**Describe alternatives you've considered**

Fix it myself, for you.  This only works if you accept new feature patches.

**Affected users and/or printers**

Everyone using the default Fusion 360 mouse settings.

fieldOfView, 2020-05-08:
As a regular 3rd party contributor, I can say that especially useful pull requests that add value to a group of users like what you propose are normally merged. There is no guarantee though. Your code would be reviewed, and needs to adhere to certain code style standards. You may be asked to make changes to the code before it gets merged. If your changes make the code harder to maintain for future versions, you may be asked to refactor your changes. 

Some pull requests take longer to get reviewed and merged than others. Sometimes a small PR is merged the same day, but I have a current open PR that's been open for around 1.5 month.

If the changes are not merged into Cura for some reason, I can also help you make your changes into a plugin that is downloadable from the Marketplace.

nallath, 2020-05-08:
This would have a fairly high chance of being merged in. Especially since we were already considering making (customizable) presets of mouse controls (eg: "Cura default", "Fusion 360", "Solidworks", etc).

gitcnd, 2020-05-08:
Thanks guys - I just wanted to check first.

rburema, 2020-05-08:
Just be aware that it might take us a while to get to your eventual PR! We've been trying to up the rate of handling PR's but we're also experiencing an uptick in PR's made, so that evens out at the moment.

gitcnd, 2020-05-08:
No worries - I just needed to know there's someone at the wheel - I occasionally contribute to stuff that's neglected or is run by curious folk who refuse to accept help etc...

closed 2020-05-08 by gitcnd