Ultimaker/Cura #7713: BUG cura 4.6.1 initial temperature ignored on the hot-end when using RAFT

Bugg69s, 2020-05-08:
initial hot-end temperature is ignored when using RAFT, it goes to normal printing temperature right away 
it works normally for the bed

mahtDFR, 2020-05-08:
Can you please fill out the issue template? Otherwise we don't have much information to work with here.

I have pasted it again here for you:

**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

Ghostkeeper, 2020-05-12:
That Google Drive page keeps giving me an error that it's not redirecting properly, sorry :( I can't download the project file. Maybe it would be better to just drag the file here into Github.

I'm unable to reproduce this problem myself. Here is the project file with the settings that I tried: [initial_layer_temperatures.zip](https://github.com/Ultimaker/Cura/files/4613859/initial_layer_temperatures.zip) . I've set the initial layer printing temperature to 222 degrees (normal is 200) and the initial layer build plate temperature to 55 degrees (normal is 50). This results in the following start sequence:

```
;Generated with Cura_SteamEngine master
M140 S55
M105
M190 S55
M104 S222
M105
M109 S222
M82 ;absolute extrusion mode
...
```

So it starts with the correct temperatures.

Then when it's about to go to layer -6 (first layer was -7):
```
G0 F9000 X136 Y128.036
M104 S200
G1 F1125 X136 Y106.969 E73.55812
... (several more extrusion and travel moves) ...
G0 F9000 X100.276 Y104.521
;TIME_ELAPSED:60.307740
;LAYER:-5
M140 S50
;TYPE:SUPPORT
G1 F1125 X99.801 Y105.3 E77.59567
```

So it starts cooling the nozzle slightly before the end of the layer, but cools the build plate only at the exact end of the layer. This is a little bit inconsistent but currently expected since the nozzle temperature has pre-cooling implemented but Cura has no information on how quick the build plate cools down.

Bugg69s, 2020-05-13:
F, sorry :/ I was changing the 
 "Initial Printing Temperature"
 instead of 
 "Printing Temperature Initial layer"

closed 2020-05-13 by Bugg69s

Ghostkeeper, 2020-05-15:
Yeah I confuse those as well sometimes. Could be named a bit more distinctly, although I can't think of a better name immediately.