Ultimaker/Cura #7714: Cura won't connect to my printer in any way

ChrisYOLO, 2020-05-08:
Chris

**Application version**
latest (4.6.1)

**Platform**
RTX 2070 RYZEN 2700X Windows 10

**Printer**
This is a custom printer i got that has a CH340 MSK Robin RepRap with 3.5 inch lcd touchscreen. In cura i was told to use Prusha i3.

**Reproduction steps**
1. reinstalled cura
2. changed com portts

**Screenshot(s)**
I have non but in the ultimaker cura monitor tab i get no information and connecttion to the printer what so ever 

**Actual results**
none i still try to connect my printer. I have changed com and baud rate. Other programs like pronterface were able to connect to my printer using com port 8 and baud rate 115200 but cura has no connection.

**Expected results**
get it working....

**Project file**
i have non as i just got the printer 

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/4597968/cura.log)

**Additional information**
Printer is brand new and the guy that sold it to me was able to use it with his cura on his computer

fieldOfView, 2020-05-08:
A workaround is to save gcode from Cura and to use Pronterface to print.

mahtDFR, 2020-05-08:
Ah yes, your log confirms this. Are there any other devices on your com ports that could be interfering?

`2020-05-08 11:48:50,520 - DEBUG - [Thread-3] USBPrinting.AutoDetectBaudJob.run [50]: Checking COM8 if baud rate 115200 works. Retry nr: 0. Wait timeout: 3
2020-05-08 11:48:55,062 - DEBUG - [Thread-3] USBPrinting.AutoDetectBaudJob.run [50]: Checking COM8 if baud rate 250000 works. Retry nr: 0. Wait timeout: 3
2020-05-08 11:49:05,586 - DEBUG - [Thread-3] USBPrinting.AutoDetectBaudJob.run [50]: Checking COM8 if baud rate 500000 works. Retry nr: 0. Wait timeout: 3
2020-05-08 11:49:16,111 - DEBUG - [Thread-3] USBPrinting.AutoDetectBaudJob.run [50]: Checking COM8 if baud rate 230400 works. Retry nr: 0. Wait timeout: 3
2020-05-08 11:49:26,634 - DEBUG - [Thread-3] USBPrinting.AutoDetectBaudJob.run [50]: Checking COM8 if baud rate 57600 works. Retry nr: 0. Wait timeout: 3
2020-05-08 11:49:37,158 - DEBUG - [Thread-3] USBPrinting.AutoDetectBaudJob.run [50]: Checking COM8 if baud rate 38400 works. Retry nr: 0. Wait timeout: 3
2020-05-08 11:49:47,685 - DEBUG - [Thread-3] USBPrinting.AutoDetectBaudJob.run [50]: Checking COM8 if baud rate 19200 works. Retry nr: 0. Wait timeout: 3
2020-05-08 11:49:58,212 - DEBUG - [Thread-3] USBPrinting.AutoDetectBaudJob.run [50]: Checking COM8 if baud rate 9600 works. Retry nr: 0. Wait timeout: 3
2020-05-08 11:50:22,223 - DEBUG - [Thread-3] USBPrinting.AutoDetectBaudJob.run [50]: Checking COM8 if baud rate 115200 works. Retry nr: 1. Wait timeout: 15
2020-05-08 11:50:42,244 - DEBUG - [Thread-3] USBPrinting.AutoDetectBaudJob.run [50]: Checking COM8 if baud rate 250000 works. Retry nr: 1. Wait timeout: 15
2020-05-08 11:51:05,278 - DEBUG - [Thread-3] USBPrinting.AutoDetectBaudJob.run [50]: Checking COM8 if baud rate 500000 works. Retry nr: 1. Wait timeout: 15
2020-05-08 11:51:28,313 - DEBUG - [Thread-3] USBPrinting.AutoDetectBaudJob.run [50]: Checking COM8 if baud rate 230400 works. Retry nr: 1. Wait timeout: 15
2020-05-08 11:51:48,342 - DEBUG - [Thread-3] USBPrinting.AutoDetectBaudJob.run [50]: Checking COM8 if baud rate 57600 works. Retry nr: 1. Wait timeout: 15
2020-05-08 11:52:11,378 - DEBUG - [Thread-3] USBPrinting.AutoDetectBaudJob.run [50]: Checking COM8 if baud rate 38400 works. Retry nr: 1. Wait timeout: 15
2020-05-08 11:52:34,414 - DEBUG - [Thread-3] USBPrinting.AutoDetectBaudJob.run [50]: Checking COM8 if baud rate 19200 works. Retry nr: 1. Wait timeout: 15`

ChrisYOLO, 2020-05-08:
i only have my regural devices camera mouse keyboard( my keyboard uses 2 usb) and a external 42+ sound card so i can use my external micorphone

mahtDFR, 2020-05-08:
Thanks for confirming. What if you use a custom FFF printer profile instead of the Prusa i3 profile?

ChrisYOLO, 2020-05-08:
i can't get any connection thats my proble. It doesn't matter what profile i use monitor ttab will remain blank

Ghostkeeper, 2020-05-12:
It's actually getting a `PermissionError` when trying to access the COM port. From your log file:

```
2020-05-07 22:38:48,077 - WARNING - [Thread-3] UM.Logger.logException [106]: Exception: Unable to create serial
2020-05-07 22:38:48,093 - WARNING - [Thread-3] UM.Logger.logException [110]: Traceback (most recent call last):
2020-05-07 22:38:48,101 - WARNING - [Thread-3] UM.Logger.logException [110]:   File "C:\Program Files\Ultimaker Cura 4.6\plugins\USBPrinting\AutoDetectBaudJob.py", line 54, in run
2020-05-07 22:38:48,105 - WARNING - [Thread-3] UM.Logger.logException [110]:     serial = Serial(str(self._serial_port), baud_rate, timeout = read_timeout, writeTimeout = write_timeout)
2020-05-07 22:38:48,108 - WARNING - [Thread-3] UM.Logger.logException [110]:   File "serial\serialwin32.py", line 31, in __init__
2020-05-07 22:38:48,112 - WARNING - [Thread-3] UM.Logger.logException [110]:   File "serial\serialutil.py", line 240, in __init__
2020-05-07 22:38:48,116 - WARNING - [Thread-3] UM.Logger.logException [110]:   File "serial\serialwin32.py", line 62, in open
2020-05-07 22:38:48,119 - WARNING - [Thread-3] UM.Logger.logException [110]: serial.serialutil.SerialException: could not open port 'COM7': PermissionError(13, 'Access is denied.', None, 5)
```

Perhaps something else is running that could be communicating to the printer? Like a different Cura instance? Or a different slicer? Or something that uploads firmware, like the Arduino IDE?

closed 2020-06-19 by no-response[bot]