Ultimaker/Cura #7717: Idea for tiny pointy thing

V1an, 2020-05-08:
Hello,
I am able, thanks to cura to get really satisfying prints,
But the only things where it is not quite perfect (yet I hope) is about tiny pointy objects.

As the layers are really shorts, they aren't able to cool down properly for the next. (As a result, layers will wobble a tiny bit)
I was wondering if CURA could allow a certain amount of time to wait before proceeding to the next layer if it has just been printed.

I'm sorry if this exist already (I couldn't find it) or if this as already been ask.
Thanks a lot to all !

fieldOfView, 2020-05-08:
Are you aware of the setting "Minimum Layer Time"?

Ghostkeeper, 2020-05-12:
Yes, see the [Minimum Layer Time](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/cooling/cool_min_layer_time.md) and perhaps the [Lift Head](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/cooling/cool_lift_head.md) setting.

closed 2020-05-12 by Ghostkeeper