Ultimaker/Cura #7719: Zoom toward mouse direction in Orthographic mode of camera rendering.

st4szek, 2020-05-08:
Please enable "zoom toward mouse direction" in orthographic mode of camera rendering.

Gr8rPwr, 2020-05-09:
This option hasn't been working for me after the update to Cura 4.6, but it worked fine in 4.4 and  4.5.  I've tried turning it off, restarting, turning it back on, restarting and it still doesn't work no matter what I do.  Am I missing something?  I loved this feature when it worked in 4.5!

Ghostkeeper, 2020-05-12:
This was disabled because zooming towards the mouse direction works through changing the position of the camera origin ("dolly" mode), but that doesn't zoom things in orthographic mode due to the nature of the orthographic projection; things that are closer are not larger. In the orthographic projection we zoom by making the view angle smaller.

tsonis, 2020-07-20:
Still, it's a basic feature for most (if not all) 3d programs, and it's pretty useful to have.

BenMarolt, 2020-08-04:
That always kinda bugged me that the camera zoom in perspective view changed the camera distance. Zooming by changing the view angle seems a lot more natural to me to use. I'm sure there are users that prefer it that way, but it would really be nice to have the zoom toward mouse direction option in orthographic mode. Pressing the scroll wheel in to reposition is not that hard, but it does feel like something extra, since it can be handled in all my other 3d modeling software. If this alternate "zoom toward mouse" is to be implemented, I'd suggest also adding an "only zoom by view angle" option along with it to give the chance to use the new implementation in both perspective and ortho.

yepher, 2020-12-31:
I would really like it if the zoom followed the mouse. Currently, the zoom seems broken to me because it never zooms to a point I am personally interested in. I usually have to zoom then pan/tilt the view multiple times to see what I actually want to see.

It would actually be nice if there were a selection on the left that was just bed manipulation so dragging pans/tilts and scroll zooms the print bed.

[image]

Ghostkeeper, 2021-01-11:
This commit disabled the checkbox when the camera is in Orthographic mode: https://github.com/Ultimaker/Cura/commit/7f559987a03a8c968cb3db9a468b8931013adfa3

You can try to remove the `enabled` property and that `Connections` element to not disable the checkbox, but I think it's still not going to work then. We disabled it for a reason. If you'd like to take a hand at fixing the issue you're welcome to try though.

Dragging the middle mouse button pans the camera. Dragging the right mouse button tilts the camera. Scrolling zooms in or out. That's what I use, anyway.